ETLCPP/etl #145: etl::delegate should does nothing if it is uninitialised

huming2207, 2019-09-05:
Hi there,

Firstly, thanks for your great library. Everything looks good for me so far.

But there's a problem. If I do something like this:

```cpp
class Foo 
{
  public:
    void call_bar() 
    {
      bar();
    };
    void assign_bar(const etl::delegate<void()>& _bar)
    {
      bar = _bar;
    };
  private:
    etl::delegate<void()> bar;
}
```

If the `bar` is not assigned with any functions by `assign_bar()` and called directly by `call_bar()` then `bar` cause my program to crash and jumps to `HardFault_Handler`.

What I think the `etl::delegate` should do, is that if it's not assigned/initialise to a function and it gets called by someone else, it should do nothing instead of crashing. Anyway, I may be wrong. Please point me out if I misunderstand something.

Regards,
Jackson

jwellbelove, 2019-09-05:
It all comes down to the perfomance/safety tradeoff. If a check is made for a 'null' delegate then there is a slight peromance hit for what is, more often than not, an initialised object. As delegates may be used in such things as a HAL, directing interrupt vectors, most people would not want a mandatory check that, in shipping code, does nothing but use up time. In the ETL I have tended towards the performance side and make most checking a responsibility of the caller. Then, error or out-of-bounds checking may be omitted when the caller knows that the call is safe (such as iterating through a vector of initialised objects).

jwellbelove, 2019-09-05:
I could add an ETL_ASSERT for the condition though.

huming2207, 2019-09-05:
@jwellbelove Yea, a static assertion is a better idea...

jwellbelove, 2019-11-04:
ETL_ASSERT added 14.35.3

closed 2019-11-04 by jwellbelove