ETLCPP/etl #147: unordered_map doesn't work with enum as a key type

mrdimfox, 2019-09-13:
Simple code for a testing:
```
enum class Test
{

};

etl::unordered_map<Test, int, 12> _;
```
and we got a compile error:
```[build] In file included from ...:
[build] ../src/etl/include\etl/unordered_map.h:1338:12: error: implicit instantiation of undefined template 'etl::hash<Test>'
[build]     hasher key_hash_function;
[build]            ^
[build] ../src/etl/include\etl/unordered_map.h:1392:32: note: in instantiation of template class 'etl::iunordered_map<Test, int, etl::hash<Test>, std::equal_to<Test> >' requested here
[build]   class unordered_map : public etl::iunordered_map<TKey, TValue, THash, TKeyEqual>
[build]                                ^
[build] ...: note: in instantiation of template class 'etl::unordered_map<Test, int, 12, 12, etl::hash<Test>, std::equal_to<Test> >' requested here
[build]     etl::unordered_map<Test, int, 12> _;
[build]                                       ^
[build] ../src/etl/include\etl/hash.h:91:32: note: template is declared here
[build]   template <typename T> struct hash;
[build]                                ^
```

Compiler CLang 8.0.1. Same for gcc arm noneabi. ETL 14.29.3.

jwellbelove, 2019-09-13:
You need to implement a specialisation of etl::hash for your enum class.

mrdimfox, 2019-09-13:
Ok. But it library we have:
```
  template <typename TKey, typename TValue, const size_t MAX_SIZE_, const size_t MAX_BUCKETS_ = MAX_SIZE_, typename THash = etl::hash<TKey>, typename TKeyEqual = std::equal_to<TKey> >
  class unordered_map : public etl::iunordered_map<TKey, TValue, THash, TKeyEqual>
...
```

We can't set own hash nicely because of `MAX_BUCKETS_ `. We should do
```
enum class Test {};
//                       duplication of size here      btw std::hash works fine with enums
//                                            \/       \/ 
using Map = etl::unordered_map<Test, int, 12, 12, std::hash<Test>>;
```

jwellbelove, 2019-10-12:
It looks like std::hash uses std::underlying_type<Enum>. I'll see if I can add it to the ETL if the compiler supportrs C++11.

jwellbelove, 2022-09-12:
Fixed 20.34.0

closed 2022-09-12 by jwellbelove