ETLCPP/etl #151: Improve platformio support

puzrin, 2019-09-22:
PIO config is outdated and buggy. Here is description of steps, required to make things work.

## PIO config

Config is outdated and buggy. Version in registry is not updated.

1. [Here](https://github.com/puzrin/etl/commit/caa5da8ec4c4e1c87e56e7eac11fd13c228a3561) is list of minimal mandadory changes. At least, that will make library useable via git links, as dependency.
   - May be content of other fields should be improved. I did only minimal possible changes to make things buildable.

2. PIO registry crawler is frozen by outdated version number in `library.properties` (arduino config). Versions in both `library.properties` AND `library.json` MUST be updated and be in sync with `include/etl/version.h`
   - After config updated, at least one new tagged release required, to kick PIO crawler.
   - I strongly recommend add CI test to make sure versions in all files are in sync.

3. IMO, announcing package in pio registly as "Embedded Template Library" is "too arduinish" :). I'd suggest change field "name" to "etl". Did not updated this in my commit, because don't know your opinion. IMO [this line](https://github.com/puzrin/etl_check/blob/master/platformio.ini#L21) in app config looks more nice than long name with space and letters of different sizes.

## PIO demo

https://github.com/puzrin/etl_check - something like this should be added into examples. Users should see, everything is buildable in PIO.

Note, `src/` folder contains garbage. I just needed something to make sure build pass and my fixes of etl config are correct. Required files from `include/` are specially done to reuse existing configs with minimal intrusion.

`platformio.ini` temporary refers to cloned repo until mainstream updated. If you copy it to mainstream's samples - don't forget to fix.

I'm not C programmer. So i can't make good code samples, but can care about fixing PIO issues. Use refered repo as template.

## Dev notes

If you need to test library config changes - clone repo to `lib/` folder of demo AND remove `lib_deps` from demo config (`platform.ini`). All subfolders of `lib/` are automatically used as dependencies.

---

@jwellbelove what do you think? I'm interested to get etl useable with pio asap. Of cause, i could use fork with fixes, but fixed mainstream is preferable.

I did not created PR because some of described changes need to be done at your side, and some depend on your opinion.

puzrin, 2019-09-22:
https://github.com/puzrin/etl_check/commit/581a705638ac4c4dda22b1650ac4963ff49a3b66

Added readme and config comments. I think, demo is ready to be included into `examples/` folder (after other fixes). May be anyone more experienced could improve `main.cpp` content, but that's not critical.

ristomatti, 2019-09-23:
> IMO, announcing package in pio registly as "Embedded Template Library" is "too arduinish" :). I'd suggest change field "name" to "etl". Did not updated this in my commit, because don't know your [...]

@puzrin I can't reply for @jwellbelove but since when has "embedded" as a term given the association to Arduino? I rarely see the terms used together.

I also don't understand how does showing that the library works on a full blown desktop computer demonstrate how it will then work for all the platforms supported by PlatformIO? Would you bet all these platforms have C++ and a version of STL compatible with ETL available:

Aceinna IMU, Atmel AVR, Atmel SAM, Espressif 32, Espressif 8266, Freescale Kinetis, GigaDevice GD32V, Infineon XMC, Intel ARC32, Intel MCS-51 (8051), Kendryte K210, Lattice iCE40, Maxim 32, Microchip PIC32, Nordic nRF51, Nordic nRF52, NXP LPC, RISC-V GAP, Samsung ARTIK, SiFive, Silicon Labs EFM32, ST STM32, ST STM8, Teensy, TI MSP430, TI TIVA, WIZNet W7500

Personally I feel much of the appeal to PlatformIO is that things "just work" even if I can see also the point in not restricting the library to certain platforms. I just hope the repo is not going to see a flood of new issues from people having issues getting it to work with PIO.

puzrin, 2019-09-23:
@ristomatti IMO you try to solve not existing problem. Alternative is to have problems right now due missed platforms/frameworks entries in list. I say so, because my projects are affected. If someone reports problem - we can decide what to do. But until that happens (and i hope it will not) - no reasons to worry.

> I also don't understand how does showing that the library works on a full blown desktop computer demonstrate how it will then work for all the platforms supported by PlatformIO?

We should not prove that every PIO's toolchain works. Goal is to show how to configure project, and demonstrate correctness of provided recipe. `native` is the most simple way to do so.

ristomatti, 2019-09-23:
@puzrin When reading https://docs.platformio.org/en/latest/what-is-platformio.html I get the picture embedded development is PlatformIO's main target area. Arduino and MBED are mentioned as examples. I would say native example does not make much sense (it doesn't hurt either of course).

If you take a look at PlatformIO libraries https://platformio.org/lib/search?query=  (they are sorted by download count). Please note that currently:
- 50 of the most downloaded libraries are for Arduino
- 9/10 of the top 10 downloaded libraries have a specific set of platforms defined

puzrin, 2019-09-24:
In embedded development GUI prototyping is very time consuming thing. If you have desktop build for gui only, that simplifies your life significantly. So, `native` may be very important part of development process.

Here is an example: https://github.com/littlevgl/lv_platformio.

I just did the most simple possible example, with zero requirements to experiment and check ETL code. You can suggest one more demo if you wish.

puzrin, 2019-09-25:
@jwellbelove , now critical things done (14.31.2 in PIO registry should work). 

Do you wish me to PR https://github.com/puzrin/etl_check as `examples/platformio`? If not - i will close issue as solved.

PS. Please, consider add CI test to check versions consistency between files. When i proposed this, meaned exactly what happened with 2 last releases :)

jwellbelove, 2019-09-25:
Yes, do the PR
I'll look into how the the CI test could be done, but I'm quite busy with a house renovation, which is taking a lot of my spare time at the moment :-(

puzrin, 2019-09-26:
Close as solved. Thanks for nice lib!

closed 2019-09-26 by puzrin