ETLCPP/etl #152: Cannot resize etl::vector composed of etl::vector

morannen, 2019-09-25:
Hi everyone,

I'm currently discovering the ETL library which looks very nice to me!
I'm struggling a little with a few things.

My objective is to have a class which has the following member :

`etl::vector<etl::vector<uint8_t, 14>, 32> mVector`

The sizes here are the maximum sizes allowed in my application.
When the class constructor is called, I would like to resize the vector with the size given as a parameter.

So I've tried the following :

```
MyClass:MyClass(uint8_t size){
    /* First resize the top level vector */
    this->mVector.resize(size);

/* Then resize and initialize each sub-vector */
    for(size_t i = 0; i < this->mVector.size();i++){
        this->mVector[i].resize(12);
        for(size_t j = 0; j < this->mVector[i].size();j++){
            this->mVector[i][j] = 0U;
        }
    }
}
```
I have the following environment : STM32F429II (ARM cortex M4), compiling with Atollic TrueStudio. When compiling, I get the following error 

`etl/vector.h:1246:36: error: call of overloaded 'move(short unsigned int&)' is ambiguous`

Also, I would like to return specific values from this big vector :
- The whole vector
- One of the sub-vectors
- Only 1 value

For example, the methods of my class could be :

```
ivector<ivector<uint8_t>> MyClass:GetFullVector(){
    return this->mVector;
}

ivector<uint8_t> MyClass:GetSubVector(uint8_t vectorIndex){
    /* Here the index is considered valid */
    return this->mVector[vectorIndex];    
}

uint8_t MyClass:GetValueFromVector(uint8_t vectorIndex, uint8_t valueIndex){
    /* Here the index is considered valid */
    return this->mVector[vectorIndex][valueIndex];    
}
```
This looks fine to me (the compiler is OK too), but I don't know how to use this on the caller side.
The result is an `etl::ivector`, which I would like to transform back into an `etl::vector` with the correct size.

Any idea?

Thanks a lot for your help!

jwellbelove, 2019-09-25:
Hi, I'll take a look, they may always be a mistake in the etl::vector code!
You can't transform an etl::ivector into an etl::vector except by creating a new one using the constructor or 'assign' (from iterator range).

jwellbelove, 2019-09-25:
I tried your code using VS2017 and GCC but I got no syntax errors.
For your class methods, you should be returning a refeference or const reference to an etl::ivector.
An etl::ivector cannot be instantiated.

jwellbelove, 2019-09-25:
etl::ivector<etl::vector<uint8_t, 14>>& GetFullVector() {
        return this->mVector;
      }

      etl::ivector<uint8_t>& GetSubVector(uint8_t vectorIndex) {
        /* Here the index is considered valid */
        return this->mVector[vectorIndex];
      }

morannen, 2019-09-25:
Thanks for your feedback!

I'm sorry, indeed I use a reference to the vectors in my code. However, could you provide me an example of how to use the result of this method? 
I'm not sure about how to recreate the vector on the caller side as the ivector might not hold the  information about the size of the original vector.

Concerning my compilation issue, I'm a bit surprised you did not have the same error! At least that means it might "just" be a compiler configuration problem. Did you use any specific GCC command? I wondered if the ambiguous overloading might be caused by a missing parameter..

jwellbelove, 2019-09-25:
If you want to create an etl::vector from a reference (preferably 'const') to an etl::ivector, the destination must be large enough to contain whatever size the ivector is.

In you exampe you have a member variable **etl::vector<etl::vector<uint8_t, 14>, 32> mVector**
The vector you copy to must be at least the same dimensions.

You could return a const reference to the internal member vector and use it as the parameter to the copy constructor. If you define the vector type in the public class definition then you could do this.

```
class MyClass
{
public:
    typedef etl::vector<etl::vector<uint8_t, 14>, 32> Container;
    // or using Container = etl::vector<etl::vector<uint8_t, 14>, 32>;

    const MyClass::Container& GetFullVector() {
        return mVector;
    }

private:

    Container mVector;
  };

MyClass myClass(5);

MyClass::Container vectorCopy(myclass);
```

jwellbelove, 2019-09-25:
Can you post the error with the ETL file and line it was complaining about?

morannen, 2019-09-25:
Thanks for the explanation!

Sure, I will post the output from the compiler tomorrow.

morannen, 2019-09-26:
So here is my code :

MyClass.h
```
#include "etl/vector.h"

class MyClass {
public:
	MyClass(size_t newSize);
	virtual ~MyClass();

private:
	etl::vector<etl::vector<uint8_t, 14>, 32> mVector;

};
``` 

MyClass.cpp
```
#include "MyClass.h"

MyClass::MyClass(size_t newSize) {
    this->mVector.resize(newSize);

    for(size_t i = 0; i < this->mVector.size();i++){
        this->mVector[i].resize(12);
        for(size_t j = 0; j < this->mVector[i].size();j++){
            this->mVector[i][j] = 0U;
        }
    }

}
```

Here is the full compilation output:

```
arm-atollic-eabi-g++ -c -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -std=gnu99 -D__weak=__attribute__((weak)) -D__packed=__attribute__((__packed__)) -DUSE_HAL_DRIVER -DSTM32F429xx -I../Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Drivers/CMSIS/Include -IC:\Users\NicolasMoranne\Documents\TestMaximDriverCPP\Drivers\STM32F4xx_HAL_Driver -O0 -ffunction-sections -fdata-sections -g -fstack-usage -Wall -mlong-calls -specs=nano.specs -o Src\main.o ..\Src\main.c 
cc1plus.exe: warning: command line option '-std=gnu99' is valid for C/ObjC but not for C++
..\Src\main.c: In function 'int main()':
..\Src\main.c:105:13: warning: unused variable 'myClassInstance' [-Wunused-variable]
   MyClass_t myClassInstance = CreateMyClassInstance(new_size);
             ^~~~~~~~~~~~~~~
arm-atollic-eabi-g++ -c ..\Src\MyClass.cpp -mthumb -mcpu=cortex-m4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -std=gnu++98 -D__weak=__attribute__((weak)) -D__packed=__attribute__((__packed__)) -DUSE_HAL_DRIVER -DSTM32F429xx -I../Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc -I../Drivers/STM32F4xx_HAL_Driver/Inc/Legacy -I../Drivers/CMSIS/Device/ST/STM32F4xx/Include -I../Drivers/CMSIS/Include -IC:\Users\NicolasMoranne\Documents\TestMaximDriverCPP\Drivers\STM32F4xx_HAL_Driver -O0 -ffunction-sections -g -fstack-usage -Wall -std=gnu++0x -fno-threadsafe-statics -specs=nano.specs -o Src\MyClass.o 
In file included from ../Inc/MyClass.h:11:0,
                 from ..\Src\MyClass.cpp:8:
../Inc/etl/vector.h: In instantiation of 'etl::vector<T, MAX_SIZE_>::vector(etl::vector<T, MAX_SIZE_>&&) [with T = unsigned char; unsigned int MAX_SIZE_ = 14u]':
../Inc/etl/vector.h:222:13:   required from 'void etl::ivector<T>::resize(size_t) [with T = etl::vector<unsigned char, 14u>; size_t = unsigned int]'
..\Src\MyClass.cpp:11:30:   required from here
../Inc/etl/vector.h:1246:36: error: call of overloaded 'move(unsigned char&)' is ambiguous
           this->push_back(std::move(*itr));
                           ~~~~~~~~~^~~~~~
In file included from c:\program files (x86)\atollic\truestudio for stm32 9.3.0\armtools\arm-atollic-eabi\include\c++\6.3.1\bits\nested_exception.h:40:0,
                 from c:\program files (x86)\atollic\truestudio for stm32 9.3.0\armtools\arm-atollic-eabi\include\c++\6.3.1\exception:171,
                 from c:\program files (x86)\atollic\truestudio for stm32 9.3.0\armtools\arm-atollic-eabi\include\c++\6.3.1\new:40,
                 from ../Inc/etl/vector.h:40,
                 from ../Inc/MyClass.h:11,
                 from ..\Src\MyClass.cpp:8:
c:\program files (x86)\atollic\truestudio for stm32 9.3.0\armtools\arm-atollic-eabi\include\c++\6.3.1\bits\move.h:101:5: note: candidate: constexpr typename std::remove_reference< <template-parameter-1-1> >::type&& std::move(_Tp&&) [with _Tp = unsigned char&; typename std::remove_reference< <template-parameter-1-1> >::type = unsigned char]
     move(_Tp&& __t) noexcept
     ^~~~
In file included from ../Inc/etl/stl/alternate/algorithm.h:42:0,
                 from ../Inc/etl/stl/algorithm.h:37,
                 from ../Inc/etl/algorithm.h:39,
                 from ../Inc/etl/vector.h:43,
                 from ../Inc/MyClass.h:11,
                 from ..\Src\MyClass.cpp:8:
../Inc/etl/stl/alternate/utility.h:153:55: note: candidate: constexpr typename etl::remove_reference<T>::type&& std::move(T&&) [with T = unsigned char&; typename etl::remove_reference<T>::type = unsigned char]
   constexpr typename etl::remove_reference<T>::type&& move(T&& t) noexcept
                                                       ^~~~

```

morannen, 2019-09-26:
What I can see is that if I comment out the first line:

`//this->mVector.resize(newSize,etl::vector<uint8_t,14>());`

everything compiles just fine, which means that for some reason the vector of vectors aspect seems to cause troubles.

jwellbelove, 2019-09-26:
Are you compiling for C++03 or C++11 or above?

morannen, 2019-09-26:
My compiler is configured for gnu++98 (C++98 + gnu extensions), however I have added a compiler argument as described in the ETL documentation : `-std=gnu++0x`

closed 2019-10-12 by jwellbelove