ETLCPP/etl #154: Cannot compile on TI Code Composer Studio

morannen, 2019-09-27:
Hi everyone,

I've been trying to compile a very simple application for a TI TMS570 using TI Code Composer Studio (version 9.0.1).

As mentioned in the ETL documentation, I defined the macro `__STDC_LIMIT_MACROS` in the project properties (I wasn't sure about which value to set though).

My application uses a very simple class called from the `sys_main.c`.

```
#include "etl/vector.h"

class MyClass {
public:
    typedef etl::vector<etl::vector<uint8_t, 14>, 32> Container;

    typedef struct{
        etl::vector<uint16_t, 14> subVector;
    }Cell_Voltage_t;

    MyClass(size_t newSize);
    etl::ivector<Cell_Voltage_t>& GetFullVector(void);

    virtual ~MyClass();

private:
    etl::vector<Cell_Voltage_t, 32> mVector;

};
```

However, since the main function is in a C file I had to implement a simple wrapper : 

```
#include "MyClass.h"

typedef void* MyClass_t;

extern "C" {
	MyClass_t CreateMyClassInstance(size_t newSize){
            return new MyClass(newSize);
        }
	
	etl::ivector<Cell_Voltage_t>& GetFullVector(MyClass_t pMyClass){
	    MyClass* pClass = static_cast<MyClass*>(pMyClass);
	    return pClass->GetFullVector();
        }
}
```

And finally, sys_main.c

```
#include "MyClassWrapper.h"

int main(void)
{
    /* USER CODE BEGIN (3) */
    /****************************************************************************
    * System Initialization
    *****************************************************************************/
    SysTick_Init();
    Wdt_Init();

    size_t new_size = 1;
    MyClass_t myClassInstance = CreateMyClassInstance(new_size);

    while(1){
    }

    return 0;
}
```

This application does not compile using TI Code Composer Studio v9.0.1. It does however compile using GCC and using Atollic TrueStudio.

- **Step 1** : Vector.h can't find <new> + some errors about methods being ambiguous.
`vector.h, line 41: fatal error #1965: cannot open source file "new"`
and 
`etl\stl\alternate\utility.h", line 98: error #268: "std::pair" is ambiguous`

[compilation_output_step1.txt](https://github.com/ETLCPP/etl/files/3662764/compilation_output_step1.txt)

After searching a little bit, I changed `#include <new>` to `#include_next <new.stdh>` to match the TI CCS directory structure. This solves the first compilation error but generates a lot more in different parts of the ETL code.

- **Step 2**: Bunch of errors in TI's `new` implementation and in ETL.

[compilation_output_step2.txt](https://github.com/ETLCPP/etl/files/3662766/compilation_output_step2.txt)

I'm pretty sure that this is due to a misconfiguration, but apart from the C (set to C99 by default) or C++ version (set to C++14 by CCS, and cannot be changed...) which can't be modified and the macro mentioned in the documentation, I can't find anything!

Anyone having a similar experience on this aspect?

Thanks!

jwellbelove, 2019-09-27:
Are you compiling with ETL_NO_STL defined, but using STL headers?
Are you able to place your 'main' function in a cpp file? That would make make 'main' a C++ function.

morannen, 2019-09-30:
Hi,

I've tried placing the main function in a cpp file and it does not seem to change anything.
Regarding the STL headers I'm not sure how to verify that. Indeed, the macro `ETL_NO_STL` is defined.

jwellbelove, 2019-09-30:
Where are you defining ETL_NO_STL? I can't see it in the command line.
What version of ARM compiler are you using ARM5 or ARM6? I think I may have found a conditional compilation bug that affects ARM6.
Can you show me your etl_profile.h?

"C:/ti/ccs901/ccs/tools/compiler/ti-cgt-arm_18.12.1.LTS/bin/armcl" -mv7R4 --code_state=32 --float_support=VFPv3D16 -O0 --include_path="C:/ti/ccs901/ccs/tools/compiler/ti-cgt-arm_18.12.1.LTS/include" --include_path="C:/ti/Hercules/F021 Flash API/02.01.01/include" --include_path="C:/Users/NicolasMoranne/workspace_ti_cc/FuckingBlinky/include" --define=__STDC_LIMIT_MACROS -g --c99 --c++14 --diag_warning=225 --diag_wrap=off --display_error_number --enum_type=packed --abi=eabi --preproc_with_compile --preproc_dependency="source/MyClass.d_raw" --obj_directory="source" --exceptions  "../source/MyClass.cpp"

morannen, 2019-09-30:
Actually I did not define my ETL_NO_STL in the command line, I thought that including the `etl_profile.h` did the job 😄 
Concerning the ARM version, my target has an ARM cortex R4-F (that is **ARMv7R**). This option does not seem to exist with ETL so I tried with ARM6.

Here is my etl_profile.h :

```
#ifndef __ETL_PROFILE_H__
#define __ETL_PROFILE_H__

#define ETL_THROW_EXCEPTIONS
#define ETL_VERBOSE_ERRORS
#define ETL_CHECK_PUSH_POP

#include "profiles/armv6_no_stl.h"
#include "ecl_user.h"

#endif

```

jwellbelove, 2019-09-30:
Ah, my bad! It is defined in "profiles/armv6_no_stl.h"
You don't need "ecl_user.h", in fact I've removed the 'C' parts of the library completely.
I think the problem is in "etl\stl\alternate\utility.h". It was not defining std::pair if the compiler was ARM6. I'm not quite sure why I added that!
I'll update the library.
BTW, I'll also make an arm7 profile with ETL_CPP14_SUPPORTED set to 1

morannen, 2019-09-30:
Great news!
Thanks for your investigation.

jwellbelove, 2019-09-30:
OK, I've pushed release 14.34.0 with the changes. See how you get on.
I'm out most of today doing a kitchen renonavtion on one of our rental houses, replacing the entire floor :-(
I shall be here for about another hour. I should be back this evening.

morannen, 2019-09-30:
Thanks! I'll try that.

Good luck with the renovation 😄

jwellbelove, 2019-10-03:
Any luck with the updates?

morannen, 2019-10-03:
Unfortunately not yet.. 
I updated yesterday TI CCS (including a compiler update) and I will check if anything has changed.

morannen, 2019-10-08:
Hi John,
Unfortunately I still cannot compile after the update of TI CCS.
Do you have any example with this IDE? I'm quite sure that there is a configuration problem with the compiler, but I cannot find it.

jwellbelove, 2019-10-08:
No, I don't have a sample TI CCS project. Do you have a simple example I can try?
I have Version: 7.4.0.00015

jwellbelove, 2019-10-08:
I see you are on a much later version. I'll try to upgrade.

morannen, 2019-10-08:
Here is a very simple project for you.
Let me know if you need any additional information for the configuration.

[ti_ccs_etl_debug_project.zip](https://github.com/ETLCPP/etl/files/3701594/ti_ccs_etl_debug_project.zip)

jwellbelove, 2019-10-09:
For some reason I couldn't get ypur project to compile as it got stuck on "Preparing for build - please wait" then fails with a configuration error, so I put together a quick test test myself.
I can see that it fails with ambiguous 'std::pair'. By changing ETL's 'pair' to a different name it would seem to be that the compiler is seeing another definition of std::pair somewhere else.
Where that somewhre else is, I haven't figured out yet.

morannen, 2019-10-09:
At least I'm glad you can reproduce the problem!
As a temporary solution, I can try to rename ETL's `pair`to something custom.
Thanks for the update!

morannen, 2019-10-10:
Indeed... 
Did you have a look at the `.stdh` files in `C:\ti\ccs910\ccs\tools\compiler\ti-cgt-arm_18.12.3.LTS\include\` and the files in `C:\ti\ccs910\ccs\tools\compiler\ti-cgt-arm_18.12.3.LTS\include\libcxx` ? I've noticed some inclusions to STD.

jwellbelove, 2019-10-10:
I think I have an idea what may be going on. A lot of files include <new>to use 'placement new'. 
In the TI compiler, at least, <new> includes <exception>, which includes <type_traits>, which appears to define std::pair.
I'm not sure what the solution is here as it seems it is very difficult to avoid the STL completely!
The only options I can see is to either create an ETL 'placement new' or put all of the 'alternate STL' functions in difference namespace.

jwellbelove, 2019-10-10:
I've just pushed a feature branch with a renamed namespace for ETL's alternate STL code and the unit tests compile fine. In my TI CCS project I've lost the std::pair error, but for some reason it wil not see the armv7.h profile in etl_profile.h, despite the additional include path being set up correctly! Very odd!

morannen, 2019-10-10:
Thanks a lot, I will give it a try!

jwellbelove, 2019-10-10:
You want 
https://github.com/ETLCPP/etl/tree/feature/change_stl_alternate_namespace

jwellbelove, 2019-10-10:
Doh!! I had arm7v.h, not armv7.h

jwellbelove, 2019-10-10:
Yay!! Compiles with no errors!

morannen, 2019-10-10:
Awesome!
Thanks a lot 😁
I will close the issue if I don't get any errors

jwellbelove, 2019-10-10:
Sorry, just realised I hadn't tested it with the arm7_no_stl.h

morannen, 2019-10-10:
It looks like the branch is missing `choose_namespace.h` in `etl/stl`. Did you push it to the feature branch?

jwellbelove, 2019-10-10:
OK, it looks like it didn't commit the new file. Will update very shortly!

jwellbelove, 2019-10-10:
OK, the feature branch now compiles for my TI CCS project for both STL and No STL profiles.

morannen, 2019-10-10:
What did you do about the missing `<new>` in `vector.h` ?

jwellbelove, 2019-10-10:
It's included in the header.

morannen, 2019-10-10:
Yes, but did you have to update the project configuration to redirect to `<new>`? Because my project cannot find this file. At some point, I considered adding `C:\ti\ccs910\ccs\tools\compiler\ti-cgt-arm_18.12.3.LTS\include\libcxx` to the project includes

jwellbelove, 2019-10-10:
No, I just installed TI CCS V9 and it just worked.

I'm using the MSP430 Compiler with these default settings...

--include_path="D:/Users/John/Documents/Programming/workspace_cc/TestETL/include" --include_path="C:/ti/ccs910/ccs/ccs_base/msp430/include" --include_path="D:/Users/John/Documents/Programming/workspace_cc/TestETL" --include_path="C:/ti/ccs910/ccs/tools/compiler/ti-cgt-msp430_18.12.3.LTS/include" --advice:power=all -g --diag_warning=225 --diag_wrap=off --display_error_number

jwellbelove, 2019-10-10:
And these linker options...

--advice:power=all -g --diag_warning=225 --diag_wrap=off --display_error_number -z -m"TestETL.map" -i"C:/ti/ccs910/ccs/ccs_base/msp430/include" -i"C:/ti/ccs910/ccs/tools/compiler/ti-cgt-msp430_18.12.3.LTS/lib" -i"C:/ti/ccs910/ccs/tools/compiler/ti-cgt-msp430_18.12.3.LTS/include" --reread_libs --diag_wrap=off --display_error_number --warn_sections --xml_link_info="TestETL_linkInfo.xml" --rom_model

morannen, 2019-10-10:
Unfortunately I get the following error :

```
etl/vector.h", line 40: fatal error #1965: cannot open source file "new"
```

It looks like the behavior of the compiler I use (ti-cgt-arm_18.12.3.LTS) is different from yours (ti-cgt-msp430_18.12.3.LTS). I will continue the investigations on my side and let you know.

Here are my default compiler settings :

```
-mv7R4 --code_state=32 --float_support=VFPv3D16 --include_path="C:/Users/NicolasMoranne/workspace_v9/DebugETL" --include_path="C:/ti/ccs910/ccs/tools/compiler/ti-cgt-arm_18.12.3.LTS/include" --include_path="C:/Users/NicolasMoranne/workspace_v9/DebugETL/include" --include_path="C:/ti/ccs910/ccs/tools/compiler/ti-cgt-arm_18.12.3.LTS/include" --define=__STDC_LIMIT_MACROS -g --relaxed_ansi --float_operations_allowed=all --diag_warning=225 --diag_wrap=off --display_error_number --enum_type=packed --abi=eabi
```

Linker (also default)

```
-mv7R4 --code_state=32 --float_support=VFPv3D16 --define=__STDC_LIMIT_MACROS -g --relaxed_ansi --float_operations_allowed=all --diag_warning=225 --diag_wrap=off --display_error_number --enum_type=packed --abi=eabi -z -m"DebugETL.map" --heap_size=0x800 --stack_size=0x800 -i"C:/ti/ccs910/ccs/tools/compiler/ti-cgt-arm_18.12.3.LTS/lib" -i"C:/ti/ccs910/ccs/tools/compiler/ti-cgt-arm_18.12.3.LTS/include" --reread_libs --diag_wrap=off --display_error_number --warn_sections --xml_link_info="DebugETL_linkInfo.xml" --rom_model
```

closed 2019-10-10 by morannen