ETLCPP/etl #156: etl::pool '-O3' optimization leads to hard fault

mrdimfox, 2019-10-08:
I have strange problem with `etl::ipool::allocate_item()` method. If I use `-O3` optimization for all project when hard fault occurs on this line of code:

```cpp
p_next = *reinterpret_cast<char**>(p_next);
```

If I turn on `-O3` optimization for everything but this two lines of code:
```cpp
uintptr_t p = reinterpret_cast<uintptr_t>(p_buffer + (items_initialised * ITEM_SIZE));
*reinterpret_cast<uintptr_t*>(p) = p + ITEM_SIZE;
```
it works great. What's can be wrong?

etl 14.34.0, gcc arm 8.3.1.

jwellbelove, 2019-10-08:
Do you know if it's faulting on a null pointer?

jwellbelove, 2019-10-08:
The code maybe running foul of the 'strict aliasing rule'. I shall investigate to see if there is another to do the calculation.

jwellbelove, 2019-10-08:
Can you try this code instead?
        
```
// Initialise another one if necessary.
if (items_initialised < MAX_SIZE)
{
  char* p = p_buffer + (items_initialised * ITEM_SIZE);
  char* np = p + ITEM_SIZE;
  *reinterpret_cast<char**>(p) = np;
  ++items_initialised;
}
```

mrdimfox, 2019-10-09:
@jwellbelove, seems like this variant works ok.

jwellbelove, 2019-10-09:
It definitely looks like a 'strict aliasing' issue. Using just char pointers gets around the problem, as these have more 'relaxed' rules relating to aliasing.

jwellbelove, 2019-10-12:
Appears fixed in 14.34.1

closed 2019-10-12 by jwellbelove