ETLCPP/etl #158: Base state in the HSM

berauser, 2019-10-09:
Firts of all thanks for the great project and the great work at the HSM!

But I wonder if it's possible to define a kind of base state that defines some additional pure virtual methods that should be implemented by all states.
For example, I would like to ask the current state if a command is allowed in the current state.

Here is a small example:
```cpp
enum class Command { DO_THIS, DO_THAT };

// The base state, that defines the pure virtual methods that should be implemented by all states
class BaseState {
	virtual ~BaseState() {}
	// the method that should be implemented by all states
	virtual bool is_allowed(Command) = 0;
}

// some magic to define the states

// For example: The Top state that implements the pure virtual method defined by BaseState
template<>
bool test::state::Top::is_allowed(Command cmd)
{
	if(cmd == Command::DO_THIS) return true;
	return false;
}

// The possible usage
Command cmd = Command::DO_THIS;
bool is_allowed = hsm.get_state()->is_allowed();
```
Is there a way to implement/realize this with HSM?

I am currently using Boost MSM to do this, but I want to replace Boost MSM.
The implementation in Boost MSM looks like this:

```cpp
struct BaseState {...}
struct TopState : public boost::msm::front::state<BaseState> {...}
```

Thanks!

jwellbelove, 2019-10-09:
I'm sure it's possible, though neither the ETL's 'fsm', 'state_chart' ort he proposed 'hsm' at the moment have one; etl::fsm will called an 'on_event_unknown' method in the derived state handler. I am not sure how useful an 'accepts' or 'is_allowed' really is, as it would likely have to scan the transition table to find out if the event will be accepted. What scenarios do you see this being use in?

berauser, 2019-10-10:
Thanks you for your quick reply

One scenario is that I want to check if I can turn off an engine in a certain situation.

For example, I can not turn off the engine when the car is moving. I have to wait until the car stops, then I can turn off the engine.

With the 'is_allowed' method I would like to ask the current state if the command is allowed at this time. If it is not allowed, then I would like to give the caller immediately a feedback. The caller must then repeat the command at a later time (when the car is stopped).

Of course I could also first query the current state and check with a list if the current command is allowed.
However, the list grows exponentially with the number of states and the commands. Likewise, the list becomes unclearly and changes become more complex.
Therefore, I want to connect the check of allowed commands with the current state (each state knows if the current command is allowed).

Unfortunately, I can not use the on_event_unkown method with the HSM because the method does not exist here and a parent state could accept the event. Furthermore I would then mix the commands (do something) with the events (something has happened). Merging would also worsen the understanding of the code and increase its complexity.

Maybe my problem can be solved differently and I just overlook something...
I'm grateful for every suggestion :)

Thank you very much.

jwellbelove, 2019-10-10:
What about guards? If transitions have guards then must they be taken into account when determining the 'allowed' state? If so, what if the guards have side effects?

closed 2020-03-28 by jwellbelove