ETLCPP/etl #160: Suggestion: Add Codacy static code analysis

escherstair, 2019-10-25:
I'm sure the source code of ETL project has a great quality, but based on my experience the usage of some tools can help the development.
One of this tools for static ocde analysis is [Codacy](https://www.codacy.com/), which can be used free of charge for [open source projects](https://opensource.codacy.com/).

Based on my experience with other GitHub projects, the integration in github project is quite easy.

If the developers think that can be useful they can setup Codacy for this github repo.

jwellbelove, 2019-10-25:
I'll definitely have a look. I'm currently on. motorcycle tour of Spain, but I'll try to set myself a reminder to check it out when I get back home.

jwellbelove, 2019-10-25:
Actually, I gave Codacy access this afternoon. The last time I looked it was still "Reviewing".

escherstair, 2019-10-28:
The first fime after you enable codacy, it takes some hours to complete its job.
Then the dasjboard with the results is available.

jwellbelove, 2019-10-28:
Just looked and it seems the ETL has an 'A'.

escherstair, 2019-10-28:
Great news!
Maybe the dashboard shows some potential issues and/or suggestion.
You can also add a caodacy badge to the readme so that everyone can reach the dashboard.

jwellbelove, 2019-10-28:
I'll have to wait til I get home to do that. I'm on the last leg of my trip, from Valladolid to catch the ferry at Santander today. Back in Plymouth tomorrow afternoon and then a ride back to Portsmouth. I'm looking forward to investigating it some more.

escherstair, 2019-10-28:
Sure :-)
The most important task now is your holiday

jwellbelove, 2019-11-04:
Added Codacy

jwellbelove, 2019-11-04:
Some files have an issue highlighted with the suggestion
_"Use -D if you want to check it. You can use -U to skip it explicitly."_
It's not clear where the -D & -U are expected to be used. Any idea?

jwellbelove, 2019-11-04:
Never mind. I used the menu on the Codacy issue page.

closed 2019-11-04 by jwellbelove