ETLCPP/etl #161: to_string(): rounding errors near int value

puzrin, 2019-10-26:
I do float value change with `0.1f` step, then `to_float()` for display. There are some expected precision drifts. Usually ETL properly rounds result. But sometime, when value is almost integer (x.999999), output is wrong:

```cpp
etl::string<15> out;
etl::to_string(2.999999f, out, etl::format_spec().precision(1));
printf("%s\n", out.c_str());
```

2.999999 => "2.10" instead of "3.0"

Could anyone confirm and fix? I use PlatformIO (with gcc).

jwellbelove, 2019-10-27:
to_string transforms floating point numbers by working on the integral and fractional parts separately. When the fraction part (0.999999) is rounded up (1.000000), it is not carrying the '1' to the integral part.
I'm in Spain at the moment, so I can't look at the issue until at least Thursday. 

If anyone else wants to have a go at a fix they are welcome to look at it, though don't forget to create a unit test that demonstrates the issue first.

jwellbelove, 2019-10-27:
A solution may be for to_string to round the floating point number before it transforms it to text. This may have other side effects!

puzrin, 2019-10-27:
AFAIK, tiny implementations of float writers usually round fractional part only and have additional check to carry overflow. I don't believe correct rounding before split can be done easy and fast.

jwellbelove, 2019-10-31:
I've updated the code to take account of roundup carry.
14.35.2

puzrin, 2019-10-31:
Tank you very much!

Tested in project, no more artefacts.

closed 2019-10-31 by puzrin