ETLCPP/etl #166: Problems with etl::vector copy constructor on Arduino

iwasz, 2019-11-21:
Dear all, I had to port some code using ETL to Arduino. I discovered that copy constructor and assignment operator for etl::vector <uint8_t, xxx> don't not work (move operations are OK). I have a fix for that which I'm ready to make a pull-request for. Tell me how to proceed. Problematic code looks like that:

```c++
using MyIsoMessage = etl::vector<uint8_t, 128>;
MyIsoMessage isoMsg;
isoMsg.push_back (1);
isoMsg.push_back (2); // Now isoMessage.size () == 2 OK

MyIsoMessage m2 ;
m2 = isoMsg; 
// m2.size () == 0 !!! 
// If we used std::move, then everything would be fine
```

My configuration is : 
* Linux Manjaro (up to date). 
* avr-gcc 9.2.0. 
* C++17 turned ON. 
* ArduinoSTL in the include path.

My etl_profile.h looks like this:

```c++
#pragma once
#define ETL_LOG_ERRORS
#define ETL_VERBOSE_ERRORS
#define ETL_CHECK_PUSH_POP

#include "etl/profiles/cpp17_no_stl.h"
```
Best regards.

jwellbelove, 2019-11-21:
Do a pull request. I'm interested to see what is going wrong with the Arduino version. Is related to using the ArduninoSTL?
BTW, if you are only using ArduinoSTL because of the ETL, there is now an option to define ETL_NO_STL, which will force the ETL to use it's own reverse engineered STL implementations.

iwasz, 2019-11-21:
Yes, I'm using ArduinoSTL solely to be able to use ETL (but as I'm porting stuff, I'll probably will be using it for other things as well). I tried ETL_NO_STL, but with it project does not compile complaining about not having ```#include <cstddef>``` in etl/include/etl/nullptr.h (line 81). There is a #if at the top which checks for : ```!defined(ARDUINO)``` which is defined in my project. 

Then I tried to shuffle macros around to enable the branch which defines ```class nullptr_t``` (line 47) but I had to remove -DARDUINO and add ETL_NO_NULLPTR_SUPPORT and ETL_COMPILER_ARM5. This helped with ```cstddef```, but was starting to look like a hack, you get the idea. But then the problem lies with : ```etl/include/etl/map.h:38``` which includes ```#include <new>```. And then I gave up.

I'll prepare the pull request right away.

jwellbelove, 2019-11-21:
Supporting the Arduino has always been problematic, as it doesn't seem to supply a lot of the library features that you would normally expect with a C++ compiler. 
I look forward to finding what the Arduino fix is. I tried your code in the unit tests and everything works fine for VC2017 & GCC
Some ETL users have got good results from using PlatformIO

iwasz, 2019-11-21:
I was surprised that there is no standard C++ library on AVR - I learned it hard way a few days ago. I created a pull request.

closed 2020-03-28 by jwellbelove