ETLCPP/etl #170: string_view/string constructors from `const char*`

rolan-reznik, 2019-11-24:
Hello,

I am using `string_view` and `string`. And it would very convenient if constructors from `const char*` be marked as explicit. I understand that stl do not have those constructors as explicit.

But maybe accepted solution would be to add some configuration macro e.g. `ETL_EXPLICIT_STRING_CONVERSION_FROM_CHAR` and mark constructors as explicit if this macro is defined?

I could make implementation of such behaviour and make a pull request if this is a good proposal?

jwellbelove, 2019-11-24:
What is the reason for this proposed difference from the STL functionality? Are you having issues with ambiguous function signatures?

rolan-reznik, 2019-11-24:
> What is the reason for this proposed difference from the STL functionality? Are you having issues with ambiguous function signatures?

Yes. For example if I want to change some method signature from `parse(const char* data)` to `parse(etl::string_view data)` all `parse` usages automatically will compile, because `const char*` will be converted to `string_view` and data length will be determined by running `strlen` (it is possible that `strlen` call will be dangerous or leads to undefned behaviour). It would be safer to manually convert all data arguments to `string_view` or `string`.

May be there are other solution. It is interesting why stl have not marked those constructors as explicit.

jwellbelove, 2019-11-24:
That sounds like a reasonable change, as long as the default behaviour is as the STL

rolan-reznik, 2019-11-24:
Ok. I will try to implement the changes and will create a pull request for `string_view` and `string`.  Is `ETL_EXPLICIT_STRING_CONVERSION_FROM_CHAR` macro name is ok?

rolan-reznik, 2019-11-24:
Of course if you are ok if I will do a pull request. :)

jwellbelove, 2019-11-24:
There will need to be two macros. One to turn on the behaviour and the other to prefix the constructor.
platform.h would have to be modified to test the first macro and define the second.

```
#if defined(ETL_ENABLE_EXPLICIT_STRING_CONVERSION_FROM_CHAR)
  #define ETL_EXPLICIT_STRING_FROM_CHAR explicit
#else
  #define ETL_EXPLICIT_STRING_FROM_CHAR
#endif
```

rolan-reznik, 2019-11-24:
`ETL_ENABLE_EXPLICIT_STRING_CONVERSION_FROM_CHAR` could be defined in `etl_profile.h`, yes?
And it will cover both `string` and `string_view`?

jwellbelove, 2019-11-24:
That is correct.
It seems logical that string and string_view should have the same behaviour.

I think I would prefer the wording ETL_FORCE_EXPLICIT_STRING_CONVERSION_FROM_CHAR as it's more explicit (no pun intended) about what it's doing.

rolan-reznik, 2019-11-24:
> I think I would prefer the wording ETL_FORCE_EXPLICIT_STRING_CONVERSION_FROM_CHAR as it's more explicit (no pun intended) about what it's doing.

Yes, I agree that it is better.

rolan-reznik, 2019-11-26:
Thank you for accepting this proposal.

closed 2019-11-27 by jwellbelove