ETLCPP/etl #182: Missing etl::string construction from etl::string_view

ArXen42, 2019-12-22:
When using STL one can write:
```cpp
std::string      first{"test"};
std::string_view first_view{first};
std::string      second{first_view};
```

However, when using ETL:
```cpp
etl::string<256> first{"test"};
etl::string_view first_view{first};
etl::string<256> second{first_view};
```
last line will fail to compile because there are no string constructor accepting string_view.

Is this problematic feature to add? Or it has design problems?

From my understanding, string_view should be universal "string pointer" (i.e. non owning representation, even if for some reason it ETL has it's own one), so it looks OK for me to be able to construct etl::string from it just like from `const char*`.

jwellbelove, 2019-12-22:
There's no reason why there can't be such a constructor. I'll add it in.

jwellbelove, 2019-12-22:
Done. 15.4.2

closed 2019-12-22 by jwellbelove

ArXen42, 2019-12-22:
Many thanks!