ETLCPP/etl #183: Trouble compiling with etl::variant and Arduino - 'variant' in namespace 'etl' does not name a type

timrbarrett, 2019-12-31:
John, 
I'm having a problem getting a variant to be recognise for compiling. I'm using Arduino. 
I tried attaching my minimal example .ino file - but it tells me the file type isn't supported. I attach it manually here.
- If I comment out the line with "variant" in it compiles without errors.
(I've included a couple of lines with a deque example to demonstrate that I can get some part of ETL working nicely with Arduino.)

Can you suggest something I might change to get a successful compile.

```
#include <Arduino.h>    
#include <ArduinoSTL.h>
#include <memory>

#include "error_handler.h"          // ETL libraries downloaded from 
                                    // https://github.com/ETLCPP/
#include "variant.h"
#include "deque.h"

void free_function(const etl::exception& e) {
  Serial.print("The error was ");   Serial.print(e.what());
  Serial.print(" in ");             Serial.print(e.file_name());
  Serial.print(" at ");             Serial.println(e.line_number());
  Serial.flush();
}
etl::error_handler::free_function callback(free_function);

void setup() {
  // put your setup code here, to run once:

  Serial.begin(115200);
  while (!Serial);
  etl::error_handler::set_callback(callback);

  typedef etl::variant<int32_t, int16_t> variantType;

  const size_t DEQUE_SIZE = 3; 
  typedef etl::deque<int, DEQUE_SIZE>    IntValueStreamDeque;

}

void loop() {
  // put your main code here, to run repeatedly:

}

The error is :
C:\Users\timrb\OneDrive\Documents\GitHub\continuous-neural-network\ETLonly\ETLonly.ino: In function 'void setup()':
ETLonly:25:11: error: 'variant' in namespace 'etl' does not name a type
   typedef etl::variant<int32_t, int16_t> variantType;
           ^
Using library ArduinoSTL in folder: C:\Users\timrb\OneDrive\Documents\GitHub\continuous-neural-network\libraries\ArduinoSTL (legacy)
Using library etl in folder: C:\Users\timrb\OneDrive\Documents\GitHub\continuous-neural-network\libraries\etl (legacy)
exit status 1
'variant' in namespace 'etl' does not name a type

jwellbelove, 2019-12-31:
I'm not sure why that's not compiling. As you are including "variant.h" then etl::variant should be visible to the compiler.

timrbarrett, 2019-12-31:
Thanks John, just making sure it wasn't something blindingly obvious. :-)

timrbarrett, 2019-12-31:
FYI: Just downloaded the full ETL libraries again, adjusted platform.h, and "private" directory. The problem continues. I'll check tomorrow if I can recreate with an UNO.

timrbarrett, 2020-01-01:
I recompiled on the UNO using Arduino UNO and Arduino/Genuino UNO and it was successful. (The device I'm trying to get the libraries working on is the now discontinued Arduino 101 based on the Curie. I guess something in the x86 libraries originally provided isn't right.) I'm not quite ready to give up.

jwellbelove, 2020-01-01:
I'm glad to hear you're making progress.
The Arduino platform has always been a bit tricky to support due to it not supporting the STL out of the box.
Some Arduino users have had success using PlatformIO as an altrnative. I've not used it in anger myself, but I hear goood things about it.

https://docs.platformio.org/en/latest/frameworks/arduino.html

timrbarrett, 2020-01-01:
I found a note on [https://www.etlcpp.com/compilers.html](url) that said "The template class variant will not currently compile unless the header file is renamed. Why?!"

I renamed variant to ETLvariant - now it compiles! You can add the Arduino101 and TinyTile to the list of boards that have the problem with variant.

jwellbelove, 2020-01-01:
Thanks for reminding me about that one. I'd completely forgotten about that glitch!

jwellbelove, 2020-01-02:
It seems that the Arduino may supply a "variant.h". As there seems to be little control over where library headers are place or what the default paths are, the Arduino's and the ETL's headers clash.
Normally in applications using the ETL the path is set to "etl/include" so that headers are included like this.
`#include "etl/variant.h"`
I don't know whether it's possible to get the Aruino platform to work like this.

timrbarrett, 2020-01-02:
If I find a way I'll document it and let you know. For now I will make use of the renamed variant capability. Thanks for this (ETL in total) and for your support.

closed 2020-01-03 by jwellbelove