ETLCPP/etl #193: Request - Scheduler example

andrewtholt, 2020-02-19:
Hi,

I'm looking for a simple example of how to use the scheduler.  Can anybody give me a hint ?

Thanks,
Andrew

jwellbelove, 2020-02-20:
The only example at the moment is that used for the scheduler unit test. It's not the simplest example that could be produced and my be a bit hard to follow.
I will look in to producing something simpler for the 'examples' directory.

As a basic example, you will have to define the following...

**Tasks**
For each of your tasks, derive a class from etl::task and overide the two virtual functions 'uint32_t task_request_work() const' and 'void task_process_work()'. 'task_request_work' returns a value that informs the scheduler that this task has work to process. The return value should be non-zero if the task has work. This meaning of this is user defined, it could be the number of messages in the task's queue, or just a 0 = no work, 1 = have work. 'task_process_work' allows the task to process any work that it has. How much work it processes on each call is user defined; often it will be one 'unit' of work and letting the policy determine which task gets the next opportunity to process more.

**Task list**
An array of pointers to the tasks. Passed to the scheduler with 'add_task_list' or 'add_task'.

**The scheduler**
An instance of etl::scheduler with the required scheduling policy (see [https://www.etlcpp.com/cooperative_scheduler.html]
Initialise the task list by calling 'add_task_list'.

**Callbacks**
If you wish to get callbacks for 'idle' or 'watchdog' then define callback functions and call 'set_idle_callback' and 'set_watchdog_callback' to tell the scheduler.
The callbacks may be global, static or member function, wrapped in an etl::function.

**Starting the scheduler**
The scheduler is started by calling 'start()'.

jwellbelove, 2020-03-29:
A simple example can be found in the repository in `examples/Scheduler`

closed 2020-03-29 by jwellbelove