ETLCPP/etl #194: Cannot compile with ETL_NO_STL due to "utility.h / algorithm.h" circularity with "etl::swap"

adfernandes, 2020-02-23:
For `ETL 16.4.2`, when `ETL_NO_STL` is defined, compilation fails with
```
In file included from etl/include/etl/algorithm.h:47,
                  from etl/include/etl/array.h:38,
                  from [...]
 etl/include/etl/utility.h:129:18: error: 'etl::swap' has not been declared
```
due to [this change](https://github.com/ETLCPP/etl/commit/9dd88e188522d27111fcd1fe2f3fc250547b7d67#diff-48d63db014c9f8b1468f7a9d343b4299R129) that adds
```
using ETL_OR_STD::swap;
```

The problem is that the top of the `algorithm.h` file **includes** `utility.h` which requires the definition of `etl::swap` that comes _after_ the inclusion of `utility.h`.

I'm happy to create a PR if @jwellbelove (or any other authority) can give me a heads-up on how they want it structured.

jwellbelove, 2020-02-23:
I can see why there should be an issue, I'm surprised that neither VS2017, GCC or CLang threw up the error on my machine! I'll pull out etl::swap into it's own header and include it in both _utility_ and _algorithm_. That should fix any issues. I'll push a new version tonight.

adfernandes, 2020-02-23:
Thanks, @jwellbelove!

FYI I'm using `Segger Embedded Studio 4.50`, with the builtin`gcc` toolchain. That toolchain gets driven at the `cc1`-level by `SES`, and it _should_ be the same as `gcc-arm-none-eabi-9-2019-q4-major`, but ... 

... with a custom compiler driver, who really knows? 🤷‍♂

jwellbelove, 2020-02-24:
I've moved etl::swap to utilities.h, which should fix the issue.

jwellbelove, 2020-02-25:
I've pushed the new code to V16.4.3
Does this solve your issue?

adfernandes, 2020-02-25:
> I've pushed the new code to V16.4.3
> Does this solve your issue?

I saw that and am _so_ appreciative, I just didn’t have time yesterday to pull and build. I’m pretty sure I can test in the next few hours!

adfernandes, 2020-02-25:
Verified that `16.4.3` fixes the issue! Thank you!

closed 2020-02-25 by adfernandes