ETLCPP/etl #198: Best way to include this in a current CMake project

ASL07, 2020-03-14:
Hi,

I want to use this library inside a CMake project.

The fastest way to do it would be to clone ETL source to my project folder and then include the  ETL headers using target_include_directories, as well as creating my etl_profile.h and including it

However, this is not really portable. Is there a way to include ETL using find_package or ExternalProject_Add?

Thanks

jwellbelove, 2020-03-15:
I can look into it, but pretty much most places I've worked didn't use CMake, so I'm not any sort of expert. The CMake files in the ETL repo were not written by me.

ASL07, 2020-03-15:
I managed to include the library as a git submodule, which seems god enough for me.
Below are the details about how I've done it. Maybe someone can suggest some improvements.

1. Add the ETL repo as a git submodule
2. On the main project CMakeLists file, download git submodules. This is needed in case the submodules are not aleady cloned (eg. if the project was cloned without recursive options)
3. Build ETL using add_subdirectory()
4. ETL include directories have to be explicitely set in the mainproject CMakeLists, I wonder why they are not automatically detected when building ETL using add_subdirectory()

This is my main project's CMakeLists.txt:
```
cmake_minimum_required(VERSION 3.0.0)
set(PROJECT_NAME "mainproject")

project(${PROJECT_NAME} VERSION 0.1.0)
set(CMAKE_CXX_STANDARD 11)

option(PACKAGE_TESTS "Build the tests" ON)
if(PACKAGE_TESTS)
    enable_testing()
    include(GoogleTest)
    add_subdirectory(tests)
endif()

# Find Git
find_package(Git QUIET)
if(GIT_FOUND AND EXISTS "${PROJECT_SOURCE_DIR}/.git")

    option(GIT_SUBMODULE "Check submodules during build" ON)
    if(GIT_SUBMODULE)
        message(STATUS "Submodule update")
        # we have a submodule setup for etl, assume it is under extern/etl
        # run this to add ETL as a submodule:
        # git submodule add https://github.com/ETLCPP/etl.git extern/etl
        # now we need to clone this submodule
        execute_process(COMMAND git submodule update --init -- extern/etl
                        WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
                        RESULT_VARIABLE GIT_SUBMOD_RESULT)

        # Throw error if there were any errors downloading the submodules
        if(NOT GIT_SUBMOD_RESULT EQUAL "0")
            message(FATAL_ERROR "git submodule update --init failed with ${GIT_SUBMOD_RESULT}, please checkout submodules")
        endif()  # if(NOT GIT_SUBMOD_RESULT EQUAL "0")  

    endif() # if(GIT_SUBMODULE)

    if(NOT EXISTS "${PROJECT_SOURCE_DIR}/extern/etl/CMakeLists.txt")
        message(FATAL_ERROR "The submodules were not downloaded! GIT_SUBMODULE was turned off or failed. Please update submodules and try again.")
    endif() # if(NOT EXISTS "${PROJECT_SOURCE_DIR}/extern/repo/CMakeLists.txt")
    
    # build etl
    add_subdirectory(extern/etl)

endif() # if(GIT_FOUND AND EXISTS "${PROJECT_SOURCE_DIR}/.git")

set(${PROJECT_NAME}_INCL_DIRS
    # Directory where my project headers are located
    # I have placed here my etl_profile.h as well
    "include/mainproject"
    # ETL headers directory
    "extern/etl/include"
)
```

Let me know your thoughts.

Btw, the library looks great, thanks for the work done!

stanczyk4, 2020-09-15:
This is what i do:

include(FetchContent)

FetchContent_Declare(
    etl
    GIT_REPOSITORY https://github.com/ETLCPP/etl.git
    GIT_TAG        18.13.0
    SOURCE_DIR     ${CMAKE_CURRENT_SOURCE_DIR}/etl
)
FetchContent_GetProperties(etl)
if (NOT etl_POPULATED)
    FetchContent_Populate(etl)
    add_subdirectory(${etl_SOURCE_DIR} ${etl_BINARY_DIR})
endif()

I run that somewhere before my target is created, then after it is i do:
target_link_libraries(${PROJECT_NAME} etl)

Thats it.

closed 2020-09-16 by jwellbelove