ETLCPP/etl #199: Unable to compile in Visual Studio

vshashi01, 2020-03-15:
Hi, 

I have been attempting to use the etl with my stm32 embedded project with VisualGDB.
I managed to resolve all the dependencies for my project. However, when I include the vector.h into my project, I see a number of errors found in the iterator.h file.

The most common error is that int is not a class error.

Whereas if I used the the array class instead, I see the error the nullptr_t is not a member of std in the file type_traits.h.

What am I doing wrong? Your help will be very helpful.

My etl_profile.h is as belows:

```
#ifndef __ETL_PROFILE_H__
#define __ETL_PROFILE_H__

#define ETL_CHECK_PUSH_POP
#define ETL_VERBOSE_ERRORS
#define ETL_NO_STL
//#define ETL_STLPORT

#include "etl/profiles/armv7_no_stl.h"

#endif // !__ETL_PROFILE_H__
```

jwellbelove, 2020-03-15:
Do you get these errors with a minimal application?

i.e.

```
#include "etl/array.h"

int main()
{
  return 0;
}
```

vshashi01, 2020-03-16:
Yes If I dont instantiate the array template class then everything compiles as expected.

jwellbelove, 2020-03-16:
Could you try setting ETL_NO_NULLPTR_SUPPORT to 1

vshashi01, 2020-03-16:
Yes! Now it compiles just as expected for the array definition!! Thanks lots. 
However, if I add etl/vector into the same file then it fails to compile again with the error "conflicting declaration typedef std::nullptr_t std::nullptr_t"

Any idea how I can resolve this?

jwellbelove, 2020-03-16:
For the Arm7 compiler ETL_NO_NULLPTR_SUPPORT should be set to 0
I'm not sure yet why vector.h can see the definition of std::nullptr and array.h can't.

vshashi01, 2020-03-16:
I see as a work around its okay for me to currently leave with array containers. I will continue digging to see if I can resolve this issue. If you do have any suggestions feel free to let me know.

jwellbelove, 2020-03-16:
I'll try to put together a minimal project to see if I get the same results, both with MSVC and VisualDBG (I think there's a trial version I can download)

vshashi01, 2020-03-16:
Oh that will be great. Meanwhile I alternatively tried to use vector.h only, with ETL_NO_NULLPTR_SUPPORT to 0, I got a different error instead.

[image]

jwellbelove, 2020-03-16:
It seems to be saying that you are calling 'assign' with 'int' parameters rather than an iterator (such as int*)

jwellbelove, 2020-03-16:
This works using the MSVC 2017 compiler

```
#define ETL_VERBOSE_ERRORS
#define ETL_NO_STL
#define ETL_THROW_EXCEPTIONS
#define ETL_DEBUG_COUNT

#define ETL_CPP11_SUPPORTED 1
#define ETL_CPP14_SUPPORTED 0
#define ETL_CPP17_SUPPORTED 0

#define ETL_NO_NULLPTR_SUPPORT                     0
#define ETL_NO_LARGE_CHAR_SUPPORT                  0
#define ETL_CPP11_TYPE_TRAITS_IS_TRIVIAL_SUPPORTED 0

#include "etl/array.h"
#include "etl/vector.h"

int main()
{
  etl::array<int, 10>  a = { 0, 1, 2, 3, 4, 5, 6, 7, 8, 9 };
  etl::vector<int, 10> v(a.begin(), a.end());

  return 0;
}
```

jwellbelove, 2020-03-16:
I've installed the trial version of VisualGDB
The only options I have for toolchains are Keil ARM5 ARMCC or ARM6 ARMClang
Are you using one of those two?

vshashi01, 2020-03-17:
I use the prebuilt Windows Toolchain for ARM with GCC9.2.1, can be found here [https://gnutoolchains.com/arm-eabi/](url)

jwellbelove, 2020-03-17:
OK, I can see what the issue is now.

nullptr.h includes <stddef.h>, which works for most compilers, but the ARM GCC9 requires that you include <cstddef>.

I used to alway use the <cxxxxx> headers in place of the <xxxx.h>. The problem was I found issues with some embedded compilers not having a <cstddef>.

I will have to add some conditional compilation to the header for GCC9

jwellbelove, 2020-03-17:
The other option is to use the global definition ::nullptr_t instead of std::nullptr

vshashi01, 2020-03-17:
Wouldnt the conditional compilation have less side effects? Thought not very maintainable?

jwellbelove, 2020-03-17:
std::nullptr is only used in about 2 or 3 places in the ETL, so the impact is minimal.
Users of the ETL don't actually need nullptr_t.

vshashi01, 2020-03-17:
Thats works for me too then :)

jwellbelove, 2020-03-18:
Should be fix in the new 16.5.2 release

closed 2020-03-18 by jwellbelove

vshashi01, 2020-03-18:
Thanks alot!!