ETLCPP/etl #203: Linker error, PPC GreenHills 2019.1.4, etl::map

ramin-raeisi, 2020-03-22:
Instantiating from etl::map passing initializer_list will lead to a linkage error as follows,

[elxr] (error #412) unresolved symbols: 1
etl::operator !=(const etl::imap<...>::iterator &, const etl::imap<...>::iterator &) from *.cpp.o

To me this is an optimization issue due to the flow of the initializer_list constructor which is different than the default constructor

jwellbelove, 2020-03-22:
Can you go into a little more detail as to what you think is happening? I don't have Greenhills and they didn't do a trial or restricted version I could download the last time I looked. What optimisation level are you using?

ramin-raeisi, 2020-03-22:
I just checked -Ospeed was set, I changed it to -Onone and it is building, will inform you eventually.

ramin-raeisi, 2020-03-22:
I forgot to mention by faking a line like 
if (obj.begin() != obj.end()) ; 
I managed to solve the issue, that is why I am certain this is an optimization issue, but interesting that changing the compiler option from -Osize to -Onone didn't help, so probably this is a linkage optimization? What you think?

jwellbelove, 2020-03-22:
Where were you inserting the fake line?

ramin-raeisi, 2020-03-22:
I put the fake line in one of the methods of my own class in which the etl::map has been declared as a member and I compared the begin and end of the map.

jwellbelove, 2020-03-22:
It seems to be saying that the linker hasn't compiled the two map iterator friend functions...
```
      friend bool operator == (const iterator& lhs, const iterator& rhs)
      {
        return lhs.p_map == rhs.p_map && lhs.p_node == rhs.p_node;
      }

      friend bool operator != (const iterator& lhs, const iterator& rhs)
      {
        return !(lhs == rhs);
      }
```

ramin-raeisi, 2020-03-22:
Exactly, but the initializer_list constructor is calling assign and respectively erase in which the != operator is being used, very strange, the assign is a template function by the way, I am also working on it.

#if ETL_CPP11_SUPPORTED && !defined(ETL_STLPORT) && !defined(ETL_NO_STL)
    //*************************************************************************
    /// Constructor, from an initializer_list.
    //*************************************************************************
    map(std::initializer_list<typename etl::imap<TKey, TValue, TCompare>::value_type> init)
      : etl::imap<TKey, TValue, TCompare>(node_pool, MAX_SIZE)
    {
      this->assign(init.begin(), init.end());
    }
#endif

jwellbelove, 2020-03-22:
Do you get the same error if you use the range constructor?

ramin-raeisi, 2020-03-22:
Let me clarify that the instance is being created using empty initializer_list like,
etl::map<....> Xxxx{};

I just found that it will be calling the default constructor not the inittializer_list constructor, right? Then if so, why removing the {} will succeed?
etl::map<....> Xxxx;

ramin-raeisi, 2020-03-22:
Confusing since the etl::map<....> Xxxx{} and etl::map<....> Xxxx seem to be the same at least based on standard C++ rules, right? Probably GHS is treating it in other way?

jwellbelove, 2020-03-23:
It's all very odd! It's beginning to look like an optimisation error in the compiler.

ramin-raeisi, 2020-03-23:
Here is a very simple test I created in which the problem exists also,

#include <iostream>
#include <etl/map.h>

class CTest {
private:
	etl::map<int, int, 2> map_{};

	void invokable()
	{
		//if (map_.begin() != map_.end());
	}

public:	
	void invoke() 
	{
		for(int i = 0; i < 100; ++i)
			invoke();
	}
};

int main()
{
	CTest ctest;
	
	ctest.invoke();
	
	std::cout << "GreenHills Compiler, etl library issue";
   
	return 0;
}

ramin-raeisi, 2020-04-12:
Conclusion, this was a known bug by GreenHills PPC 2019.1.4 and we got a patch for it.

closed 2020-04-12 by ramin-raeisi