ETLCPP/etl #210: Cannot use etl::map on Arduino AVR

labodj, 2020-04-13:
Hello,
I'm facing a problem using etl::map with Arduino AVR, I'm compiling with GCC 9.3.0 AVR (you can find it [here](https://blog.zakkemble.net/avr-gcc-builds/comment-page-1/#comment-209216)) using PlatformIO under VS Code and I'm using ArduinoSTL library support ([this](https://github.com/ciband/avr_stl)).

This is my **etl_profile.h**:
```c
#ifndef __ETL_PROFILE_H__
#define __ETL_PROFILE_H__

#define ETL_VERBOSE_ERRORS
#define ETL_CHECK_PUSH_POP
#define ETL_TARGET_DEVICE_GENERIC
#define ETL_TARGET_OS_NONE
#define ETL_COMPILER_GCC
#define ETL_CPP11_SUPPORTED 1
#define ETL_CPP14_SUPPORTED 1
#define ETL_CPP17_SUPPORTED 1
#define ETL_NO_NULLPTR_SUPPORT 0
#define ETL_NO_LARGE_CHAR_SUPPORT 1
#define ETL_CPP11_TYPE_TRAITS_IS_TRIVIAL_SUPPORTED 0
#define ETL_NO_STL

#endif
```

this is my **main.cpp**:
```c++
#include <Arduino.h>
#include <ArduinoSTL.h>
#include <etl/map.h>

void setup()
{
  etl::map<const int, int, 10> test;
  test[1] = 2;
}

void loop()
{
}
```

and this is the **error**:

> Building in release mode
> Compiling .pio/build/release/src/main.cpp.o
> In file included from .pio/libdeps/release/Embedded Template Library_ID930/include/etl/iterator.h:36,
>                  from .pio/libdeps/release/Embedded Template Library_ID930/include/etl/container.h:37,
>                  from .pio/libdeps/release/Embedded Template Library_ID930/include/etl/algorithm.h:44,
>                  from .pio/libdeps/release/Embedded Template Library_ID930/include/etl/map.h:40,
>                  from src/main.cpp:3:
> .pio/libdeps/release/Embedded Template Library_ID930/include/etl/utility.h: In instantiation of 'constexpr etl::pair<T1, T2>::pair(U1&&, U2&&) [with U1 = etl::imap<const int, int, etl::less<const int> >::iterator; U2 = bool; T1 = etl::imap<const int, int, etl::less<const int> >::iterator; T2 = bool&]':
> .pio/libdeps/release/Embedded Template Library_ID930/include/etl/utility.h:208:10:   required from 'etl::pair<T1, T2> etl::make_pair(T1&&, T2&&) [with T1 = etl::imap<const int, int, etl::less<const int> >::iterator; T2 = bool&]'
> .pio/libdeps/release/Embedded Template Library_ID930/include/etl/map.h:1142:35:   required from 'etl::pair<etl::imap<TKey, TMapped, TKeyCompare>::iterator, bool> etl::imap<TKey, TMapped, TKeyCompare>::insert(etl::imap<TKey, TMapped, TKeyCompare>::rvalue_reference) [with TKey = const int; TMapped = int; TKeyCompare = etl::less<const int>; etl::imap<TKey, TMapped, TKeyCompare>::rvalue_reference = etl::pair<const int, int>&&; etl::imap<TKey, TMapped, TKeyCompare>::value_type = etl::pair<const int, int>]'
> .pio/libdeps/release/Embedded Template Library_ID930/include/etl/map.h:925:71:   required from 'etl::imap<TKey, TMapped, TKeyCompare>::mapped_type& etl::imap<TKey, TMapped, TKeyCompare>::operator[](etl::imap<TKey, TMapped, TKeyCompare>::key_parameter_t) [with TKey = const int; TMapped = int; TKeyCompare = etl::less<const int>; etl::imap<TKey, TMapped, TKeyCompare>::mapped_type = int; etl::imap<TKey, TMapped, TKeyCompare>::key_parameter_t = const int]'
> src/main.cpp:8:9:   required from here
> .pio/libdeps/release/Embedded Template Library_ID930/include/etl/utility.h:131:63: error: cannot bind non-const lvalue reference of type 'bool&' to an rvalue of type 'bool'
>   131 |       : first(etl::forward<U1>(a)), second(etl::forward<U2>(b))
>       |                                                               ^
> *** [.pio/build/release/src/main.cpp.o] Error 1
>

jwellbelove, 2020-04-13:
There is an error in make_pair in utility.h at line 214
I will update the library, but if you want to make a quick patch to test the change, then modify line 214 to `return pair<T1, T2>(etl::forward<T1>(a), etl::forward<T2>(b));`

labodj, 2020-04-13:
Thanks for the fast reply,
the line in my case is at 208, but it seems that your fix is working as expected.

jwellbelove, 2020-04-13:
I sometimes find it tricky to remember when I should be using 'move' or 'forward'.

closed 2020-05-11 by jwellbelove