ETLCPP/etl #212: Error when wrapping for use with C

4204015, 2020-04-21:
Hello,
I want to use the ETL in a C project but get many `template with C linkage` errors when compiling with `arm-none-eabi-gcc` (v4.9, C++11), e.g.:
```
 /root/.conan/data/embedded-template-library/16.5.0/la/stable/package/5ab84d6acfe1f23c4fae0ab88f26e3a396351ac9/include/etl/type_traits.h:81:3: error: template with C linkage
    template <typename T, const T VALUE>
    ^
 /root/.conan/data/embedded-template-library/16.5.0/la/stable/package/5ab84d6acfe1f23c4fae0ab88f26e3a396351ac9/include/etl/type_traits.h:99:3: error: template with C linkage
    template <typename T, const T VALUE>
    ^
...
```

the header file:

```cpp
#ifndef TEST_WRAPPER_H
#define TEST_WRAPPER_H

#ifdef __cplusplus
extern "C" {
#endif

// Place your prototype and variable declarations here
double getMean(double values[]);

#ifdef __cplusplus
}
#endif

#endif //TEST_WRAPPER_H
```
and the source file:
```cpp
#include "cppWrapper.h"
#include "etl/vector.h"
#include <numeric>

double getMean(double values[]) {

    etl::vector<double, 5> vec;
    for (std::size_t i=0; i < 5; i++) {
        vec.push_back(values[i]);
    }

    double sum = std::accumulate(vec.begin(), vec.end(), 0.0);
    return sum / 5;
}
```

Any ideas on the problem?

jwellbelove, 2020-04-21:
Is your source file in a '.c' or '.cpp' file?

jwellbelove, 2020-04-21:
I've compiled your code in VS2019 and GCC, but I get no errors.

4204015, 2020-04-22:
It finally compiled without errors when I incorporated ETL as a submodule instead of a self-made faulty(?) conan package. I am looking forward to an official conan package release ...
Thanks for your reponse and the great lib.

closed 2020-04-22 by 4204015

jwellbelove, 2020-04-22:
I've got most of the way there to uploading the ETL to conan. I'm stuck at the last hurdle where I'm trying to execute  

`conan upload "embedded-template-library" -r embedded-template-library`

but it refuses to accept my Bintray username and password!

jwellbelove, 2020-04-22:
```
PS D:\Users\John\Documents\Programming\GitHub\etl> conan upload "embedded-template-library" -r embedded-template-library
Are you sure you want to upload 'embedded-template-library/17.4.2' to 'embedded-template-library'? (yes/no): yes
Uploading to remote 'embedded-template-library':
Uploading embedded-template-library/17.4.2 to remote 'embedded-template-library'
Please log in to "embedded-template-library" to perform this action. Execute "conan user" command.
If you don't have an account sign up here: https://bintray.com/signup/oss
Remote 'embedded-template-library' username: jwellbelove
Please enter a password for "jwellbelove" account:
ERROR: Wrong user or password
```