ETLCPP/etl #213: Using ETL in Zephyr

nandra, 2020-04-21:
Hi,

I'm trying to compile blinky application in Zyphyr OS as sample application but getting issue when would like to include <new> in list.h that file doesn't exists. Look like toolchain include path is somehow not right. Just want to ask if anyone have some experience with ETL in Zephyr. Thanks a lot.

BR,

marek

jwellbelove, 2020-04-21:
Not heard of Zephyr before, but it looks very interesting. What OS and tools are you developing with?

nandra, 2020-04-22:
Using linux and building Zephyr OS in docker container and target is stm32f091 (nucleo). I posted also issue on Zephyr github lets see. Keep you posted. Thanks

nandra, 2020-04-23:
With zephyr community help I was able to resolve an issue and push some draft here: https://github.com/nandra/zephyr/commit/428183d9f85a4cd11f2fd285ea3554622fc19aad if anyone interested. Thanks.

closed 2020-04-23 by jwellbelove