ETLCPP/etl #214: IAR and etl::delegate compilation errors

Hittherhod, 2020-04-28:
When I have the following code:
```
struct LedManager
{
    void start(unsigned a, unsigned b { ... }
};
static LedManager led;
static auto led_blink = etl::delegate<void(unsigned, unsigned)>::create<decltype(led), led, &decltype(led)::start>();
```

I get the following answers using IAR 8.40.2 (which has C++17 language features but only the C++14 standard library):
```
Error[Pa073]: invalid type conversion (from "void *" to "void (*)(void *, unsigned int, unsigned int)") C:\Users\hittherhod\MyProj\external\etl\include\etl\delegate.h 378 
          detected during: 
            instantiation of "inline TReturn etl::delegate<TReturn (TParams...)>::lambda_stub<TLambda>(void *, TParams...) [with TReturn=void, TParams=<uint32_t, uint32_t>, TLambda=void  
(void *, unsigned int, unsigned int)]" at line 113 
            instantiation of "inline etl::delegate<TReturn (TParams...)>::delegate(TLambda const &) [with TReturn=void, TParams=<uint32_t, uint32_t>, TLambda=void (void *, unsigned int,  
unsigned int)]" at line 171 
            instantiation of "inline etl::delegate<TReturn (TParams...)> etl::delegate<TReturn (TParams...)>::create<T,Instance,Method>() [with TReturn=void, TParams=<uint32_t, uint32_t>,  
T=LedManager, Instance=led_manager, Method=&LedManager::start]" at line 269 of "C:\Users\hittherhod\MyProj\src\Application.cpp" 
Error[Pe131]: expression must have pointer-to-class type C:\Users\hittherhod\MyProj\external\etl\include\etl\delegate.h 379 
          detected during: 
            instantiation of "inline TReturn etl::delegate<TReturn (TParams...)>::lambda_stub<TLambda>(void *, TParams...) [with TReturn=void, TParams=<uint32_t, uint32_t>, TLambda=void  
(void *, unsigned int, unsigned int)]" at line 113 
            instantiation of "inline etl::delegate<TReturn (TParams...)>::delegate(TLambda const &) [with TReturn=void, TParams=<uint32_t, uint32_t>, TLambda=void (void *, unsigned int,  
unsigned int)]" at line 171 
            instantiation of "inline etl::delegate<TReturn (TParams...)> etl::delegate<TReturn (TParams...)>::create<T,Instance,Method>() [with TReturn=void, TParams=<uint32_t, uint32_t>,  
T=LedManager, Instance=led_manager, Method=&LedManager::start]" at line 269 of "C:\Users\hittherhod\MyProj\src\Application.cpp" 
```

This works with arm-none-eabi-g++ 9.2.1. I can't use std::function because it isn't guaranteed not to allocate (I'm in a no-heap-allowed environment). I think the first error might just be an over-strictness in IAR's pointer casting semantics, but I don't understand the last part. The Instance is a reference and Method is a pointer to a member function, so it should satisfy those constraints.

jwellbelove, 2020-04-28:
Could you try changing line 378 in delegate.h from
`TLambda* p = static_cast<TLambda*>(object);`
to
`TLambda* p = reinterpret_cast<TLambda*>(object);`

Hittherhod, 2020-04-28:
So with that change the first error goes away, but I still get the 'expression must have pointer-to-class type' error.

jwellbelove, 2020-04-28:
What is the error output you are getting now?

Hittherhod, 2020-04-28:
```
Error[Pe131]: expression must have pointer-to-class type C:\Users\hittherhod\MyProj\external\etl\include\etl\delegate.h 379 
          detected during: 
            instantiation of "inline TReturn etl::delegate<TReturn (TParams...)>::lambda_stub<TLambda>(void *, TParams...) [with TReturn=void, TParams=<uint32_t, uint32_t>, TLambda=void  
(void *, unsigned int, unsigned int)]" at line 113 
            instantiation of "inline etl::delegate<TReturn (TParams...)>::delegate(TLambda const &) [with TReturn=void, TParams=<uint32_t, uint32_t>, TLambda=void (void *, unsigned int,  
unsigned int)]" at line 171 
            instantiation of "inline etl::delegate<TReturn (TParams...)> etl::delegate<TReturn (TParams...)>::create<T,Instance,Method>() [with TReturn=void, TParams=<uint32_t, uint32_t>,  
T=LedManager, Instance=led_manager, Method=&LedManager::start]" at line 269 of "C:\Users\hittherhod\MyProj\src\Application.cpp" 
```

jwellbelove, 2020-04-28:
Could you try changing the other static_cast instances to reinterpret_cast? (320 & 330)

Hittherhod, 2020-04-28:
I still get the same error even with those changes. I'm not sure why the reinterpret_cast is still placing this restriction. I thought it should be discarded with the "trust me" of reinterpret_cast.

jwellbelove, 2020-04-28:
Well, It was a bit of a long shot, but I thought I'd try that first.

The error seems to be saying that it thinks `p` is not a pointer to a class.

```
TLambda* p = reinterpret_cast<TLambda*>(object);
return (p->operator())(arg...); // Error[Pe131]: expression must have pointer-to-class type
```

jwellbelove, 2020-04-28:
From the error, it seems to be seeing TLambda as an ordinary function pointer.

`TLambda=void (void *, unsigned int, unsigned int)`

jwellbelove, 2020-04-28:
Could you try commenting out the functions at lines 110, 128, 213, 375. (All the Lambda ones)

Hittherhod, 2020-04-28:
That finally lets it compile. So I guess it's something to do with IAR's overload resolution algorithm?

Hittherhod, 2020-04-28:
So, I have no idea why, but if I change the function at line 110 to be:
```
template <typename TLambda, typename = typename std::enable_if<std::is_class<TLambda>::value>::type>
    delegate(const TLambda& instance)
    {
      assign((void*)(&instance), lambda_stub<TLambda>);
    }
```
then it works (even with the others no longer commented out).

jwellbelove, 2020-04-28:
It looks like it must be something like that. I'm not sure why it's trying to match with the lambdas as they only accept one template parameter, and not three like you are using. 
I may have to put a compilation guard in to disable lambdas for IAR. :-(

jwellbelove, 2020-04-28:
Sorry, posted before your message came through.
I'll have to implement `is_class` in the ETL to do that as the library must work without STL support.

Hittherhod, 2020-04-28:
Possibly, all though the SFINAE method I posted works with C++11, so maybe that's all that's needed. I don't know how to check out the overload set with IAR. G++ was printing them out when it failed to match which was really nice.

jwellbelove, 2020-04-28:
I'll knock together an ETL version of is_class and update the code.

Hittherhod, 2020-04-28:
Ah, yeah, that's unfortunate. Well, I have a path forward for this, so should I close the ticket or wait for the update and have that PR close it?

jwellbelove, 2020-04-28:
Carry on with your modded code and I'll update the library.
I'll close the issue when I push to master.

closed 2020-05-07 by jwellbelove