ETLCPP/etl #215: Invalid operands to binary expression 'nullptr == etl::optional'

Hittherhod, 2020-05-06:
So, I'm using an external library that wants to check `etl::optional` against a nullptr for equality, but it fails with a `invalid operands to binary expression` error (in this specific case, it is a populated `etl::optional<std::uint16_t>` being compared and the error is that it is trying to compare `std::uint16_t` and `nullptr`. 

With`std::optional`, this doesn't happen, even though `etl::optional` looks to have compatibility as per cppreference. I'd use `std::optional` but I need to support IAR which does not have C++17 library features.

jwellbelove, 2020-05-06:
I'll take a look ASAP.
It's odd that it is not getting caught by operator ==, conditions 19 or 20.

jwellbelove, 2020-05-06:
I tried `std::optional` and I can't get it to compile with `nullptr`.

```
std::optional<std::uint16_t> opt(1U);

// Error C2679	binary '==': no operator found which takes a right-hand operand of type 'nullptr' (or there is no acceptable conversion)
bool b1 = (opt == nullptr); 

// Error C2679	binary '==': no operator found which takes a left-hand operand of type 'nullptr' (or there is no acceptable conversion)
bool b2 = (nullptr == opt);
```

jwellbelove, 2020-05-06:
Did you mean `std::nullopt`?

Hittherhod, 2020-05-06:
So this is with the [Trompeloeil](https://github.com/rollbear/trompeloeil) mocking library. So I try to mock a return value of `etl::optional<std::uint16_t>` but I see that it's trying to instantiate the following function template:

```
      in instantiation of function template specialization 'etl::operator==<unsigned short,
      nullptr_t>' requested here
    return t == nullptr; // From this point here
```

But it's completely okay when I use an `std::optional` instead. It looks like it's using some custom type traits:

```c++
using tag = std::integral_constant<bool, is_null_comparable<T>::value
                                       && !is_matcher<T>::value
                                       && !std::is_array<T>::value>;
```
with `is_null_comparable` being defined as

```
  template <typename T, typename U>
  using equality_comparison = decltype((std::declval<T const&>() == std::declval<U const&>())
                                       ? true
                                       : false);

  template <typename T, typename U>
  using is_equal_comparable = is_detected<equality_comparison, T, U>;

template <typename T>
  using is_null_comparable = is_equal_comparable<T, indirect_null>;
```

I'm on a Mac compiling these (as unit tests), so indirect_null is just a struct with a `operator std::nullptr_t();` declaration.

So this library seems to think they are indeed comparable. If I change the mocked return value to `std::optional<std::uint16_t>` I don't see this problem.

jwellbelove, 2020-05-06:
Well, I've tried comparing `std::optional` with `nullptr` with MSVC, GCC and CLang on Windows and GCC on Linux, and none of them will accept it as valid.

There are only two equality operators that could match with comparison with `nullptr`, according to cppreference.com

```
template< class T, class U >constexpr bool operator==( const optional<T>& opt, const U& value); 
template< class T, class U >constexpr bool operator==( const T& value, const optional<U>& opt );
```
When they compare values they try to compare `uint16_t` with `nullptr`, which is not valid.

Hittherhod, 2020-05-06:
So the library that’s causing this error seems to use some SFINAE tricks to
decide whether the type can be compared to a nullptr. I noticed that the
libc++ on my machine has some SFINAE trickery around the operator == too,
so maybe that convinces the other checker that it cannot be compared. I’ll
try adding those constraints in on my local copy to see if that works.

Hittherhod, 2020-05-06:
Yup, that's it. If I change 19 and 22 to be
```
template <typename T, typename U, 
typename = std::enable_if_t<std::is_convertible_v<decltype(std::declval<T>() == declval<U>), bool>>
>
```
then my problem disappears. This is only possible with C++11 or later (I'm just lazy and use _t and _v whenever). There might be a way to do this that's C++03 compatible, but it's probably rare enough to not be with it.

jwellbelove, 2020-05-07:
I tried those changes and they don't work for me, and in fact causes other previously working unit tests to fail to compile.

Which operator== overload is your code calling for the parameter pair `nullptr` and `etl::optional<std::uint16_t>`

Hittherhod, 2020-05-07:
> On my compilers decltype(std::declval<T>() == std::declval<U>()) is a
> compilation error for T = std::uint16_t and U = std::nullptr_t, which
> what I would have expected.
>
> Error E0042 operand types are incompatible ("uint16_t" and
> "std::nullptr_t")

I had it as an extra anonymous third template parameter. That way it will
fail when doing later SFINAE checks for whether that is valid syntax or
not. I think I forgot to copy the whole function. I just copied the changed
template list.

jwellbelove, 2020-05-07:
So are you able to tell me which `etl::optional` `operator==` your code is calling, as AFAIK `optional<std::uint16>` and `nullptr`  should not be valid parameters for _any_ of the supplied `operator==`. 
cppreference.com only states that std::nullopt is comparable with an `optional`.

Hittherhod, 2020-05-07:
So the external library I was using was reaching 19 because it did it’s own
check and saw that etl::optional had an operator== that accepts nullptr
(since there is no constraint on U or T) so it tried to check instead of
just saying no.

jwellbelove, 2020-05-07:
Ok, I see.

closed 2020-05-07 by Hittherhod