ETLCPP/etl #217: Test Compilation Errors with Clang

phillipjohnston, 2020-05-06:
When I compile the unit tests with GCC, all test files compile successfully. When I compile with Clang (currently using OS X Clang), the following files fail to compile with a consistent error pattern:

```
'test/test_deque.cpp',
'test/test_flat_map.cpp',
'test/test_flat_multimap.cpp',
'test/test_flat_multiset.cpp',
'test/test_flat_set.cpp',
'test/test_map.cpp',
'test/test_multimap.cpp',
'test/test_multiset.cpp',
'test/test_reference_flat_map.cpp',
'test/test_reference_flat_multimap.cpp',
'test/test_reference_flat_multiset.cpp',
'test/test_reference_flat_set.cpp',
'test/test_pearson.cpp',
'test/test_set.cpp',
'test/test_string_view.cpp',
'test/test_to_u16string.cpp',
'test/test_to_u32string.cpp',
'test/test_to_string.cpp',
'test/test_to_wstring.cpp',
'test/test_unordered_map.cpp',
'test/test_unordered_multimap.cpp',
'test/test_unordered_multiset.cpp',
'test/test_unordered_set.cpp',
```

This is the primary error pattern:

```
In file included from ../test/test_flat_multimap.cpp:29:
In file included from /usr/local/Cellar/unittest-cpp/2.0.0/include/UnitTest++/UnitTest++.h:1:
In file included from /usr/local/Cellar/unittest-cpp/2.0.0/include/UnitTest++/UnitTestPP.h:6:
In file included from /usr/local/Cellar/unittest-cpp/2.0.0/include/UnitTest++/CheckMacros.h:6:
/usr/local/Cellar/unittest-cpp/2.0.0/include/UnitTest++/Checks.h:24:32: error: invalid operands to binary expression ('basic_ostream<char, std::__1::char_traits<char> >' and 'const etl::ireference_flat_multimap<int, TestDataNDC<std::__1::basic_string<char> >, etl::less<int> >::const_iterator')
         stream << "Expected " << expected << " but was " << actual;
         ~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~~~~~
../test/test_flat_multimap.cpp:437:7: note: in instantiation of function template specialization 'UnitTest::CheckEqual<etl::ireference_flat_multimap<int, TestDataNDC<std::__1::basic_string<char> >, etl::less<int> >::const_iterator, etl::ireference_flat_multimap<int, TestDataNDC<std::__1::basic_string<char> >, etl::less<int> >::const_iterator>' requested here
      CHECK_EQUAL(constData.begin(), std::begin(constData));
      ^
/usr/local/Cellar/unittest-cpp/2.0.0/include/UnitTest++/CheckMacros.h:151:27: note: expanded from macro 'CHECK_EQUAL'
      #define CHECK_EQUAL UNITTEST_CHECK_EQUAL
                          ^
/usr/local/Cellar/unittest-cpp/2.0.0/include/UnitTest++/CheckMacros.h:40:17: note: expanded from macro 'UNITTEST_CHECK_EQUAL'
      UnitTest::CheckEqual(*UnitTest::CurrentTest::Results(), expected, actual, UnitTest::TestDetails(*UnitTest::CurrentTest::Details(), __LINE__)); \
                ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/ostream:218:20: note: candidate function not viable: no known conversion from 'const etl::ireference_flat_multimap<int, TestDataNDC<std::__1::basic_string<char> >, etl::less<int> >::const_iterator' to 'const void *' for 1st argument; take the address of the argument with &
    basic_ostream& operator<<(const void* __p);
                   ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/type_traits:4830:3: note: candidate function template not viable: no known conversion from 'basic_ostream<char, std::__1::char_traits<char> >' to 'std::byte' for 1st argument
  operator<< (byte  __lhs, _Integer __shift) noexcept
  ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/ostream:194:20: note: candidate function not viable: no known conversion from 'const etl::ireference_flat_multimap<int, TestDataNDC<std::__1::basic_string<char> >, etl::less<int> >::const_iterator' to 'std::__1::basic_ostream<char> &(*)(std::__1::basic_ostream<char> &)' for 1st argument
    basic_ostream& operator<<(basic_ostream& (*__pf)(basic_ostream&))
                   ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/ostream:198:20: note: candidate function not viable: no known conversion from 'const etl::ireference_flat_multimap<int, TestDataNDC<std::__1::basic_string<char> >, etl::less<int> >::const_iterator' to 'basic_ios<std::__1::basic_ostream<char, std::__1::char_traits<char> >::char_type, std::__1::basic_ostream<char, std::__1::char_traits<char> >::traits_type> &(*)(basic_ios<std::__1::basic_ostream<char, std::__1::char_traits<char> >::char_type, std::__1::basic_ostream<char, std::__1::char_traits<char> >::traits_type> &)' (aka 'basic_ios<char, std::__1::char_traits<char> > &(*)(basic_ios<char, std::__1::char_traits<char> > &)') for 1st argument
    basic_ostream& operator<<(basic_ios<char_type, traits_type>&
                   ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/ostream:203:20: note: candidate function not viable: no known conversion from 'const etl::ireference_flat_multimap<int, TestDataNDC<std::__1::basic_string<char> >, etl::less<int> >::const_iterator' to 'std::__1::ios_base &(*)(std::__1::ios_base &)' for 1st argument
    basic_ostream& operator<<(ios_base& (*__pf)(ios_base&))
                   ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/ostream:206:20: note: candidate function not viable: no known conversion from 'const etl::ireference_flat_multimap<int, TestDataNDC<std::__1::basic_string<char> >, etl::less<int> >::const_iterator' to 'bool' for 1st argument
    basic_ostream& operator<<(bool __n);
                   ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/ostream:207:20: note: candidate function not viable: no known conversion from 'const etl::ireference_flat_multimap<int, TestDataNDC<std::__1::basic_string<char> >, etl::less<int> >::const_iterator' to 'short' for 1st argument
    basic_ostream& operator<<(short __n);
                   ^
/Library/Developer/CommandLineTools/usr/include/c++/v1/ostream:208:20: note: candidate function not viable: no known conversion from 'const etl::ireference_flat_multimap<int, TestDataNDC<std::__1::basic_string<char> >, etl::less<int> >::const_iterator' to 'unsigned short' for 1st argument
    basic_ostream& operator<<(unsigned short __n);
[... 85 lines cut ...]

A secondary pattern that occurs is:

```
In file included from ../test/test_string_view.cpp:29:
In file included from /usr/local/Cellar/unittest-cpp/2.0.0/include/UnitTest++/UnitTest++.h:1:
In file included from /usr/local/Cellar/unittest-cpp/2.0.0/include/UnitTest++/UnitTestPP.h:6:
In file included from /usr/local/Cellar/unittest-cpp/2.0.0/include/UnitTest++/CheckMacros.h:6:
/usr/local/Cellar/unittest-cpp/2.0.0/include/UnitTest++/Checks.h:24:32: error: call to function 'operator<<' that is neither visible in the template definition nor found by argument-dependent lookup
         stream << "Expected " << expected << " but was " << actual;
                               ^
../test/test_string_view.cpp:386:7: note: in instantiation of function template specialization 'UnitTest::CheckEqual<etl::basic_string_view<char, etl::char_traits<char> >, etl::basic_string_view<char, etl::char_traits<char> > >' requested here
      CHECK_EQUAL(view1, sub);
      ^
/usr/local/Cellar/unittest-cpp/2.0.0/include/UnitTest++/CheckMacros.h:151:27: note: expanded from macro 'CHECK_EQUAL'
      #define CHECK_EQUAL UNITTEST_CHECK_EQUAL
                          ^
/usr/local/Cellar/unittest-cpp/2.0.0/include/UnitTest++/CheckMacros.h:40:17: note: expanded from macro 'UNITTEST_CHECK_EQUAL'
      UnitTest::CheckEqual(*UnitTest::CurrentTest::Results(), expected, actual, UnitTest::TestDetails(*UnitTest::CurrentTest::Details(), __LINE__)); \
                ^
../test/test_string_view.cpp:57:17: note: 'operator<<' should be declared prior to the call site or in namespace 'etl'
  std::ostream& operator << (std::ostream& os, const View& view)
                ^
1 error generated.
```

phillipjohnston, 2020-05-06:
I tried poking around to find a solution, but this templating problem is a bit over my head.

jwellbelove, 2020-05-06:
OK, I'll take a look. Trying to understanding template errors can be a bit puzzle, but you get the knack after a while of finding the crux issue.
I may not get much done this evening as it's already gone 7pm here. I assume it's not an urgent issue you want solved?

phillipjohnston, 2020-05-06:
Not urgent at all, especially since the same tests compile fine with gcc-9.

jwellbelove, 2020-05-06:
I test on Windows with CLang and have no issues.

phillipjohnston, 2020-05-06:
It must related to OS X's clang version, which seems to be a bit different. I'll refine the exclusion logic and check mainline clang too.

phillipjohnston, 2020-05-06:
Mainline clang on OS X also has the problem, seems related to the system's headers. Updated logic in #218 restricts it to OS X only.

jwellbelove, 2020-05-07:
Can you give this unit test file a try to see if I am on the right track?

test_flat_multimap.cpp
on branch
https://github.com/ETLCPP/etl/tree/hotfix/fix-unit-test-errors-for-mac-clang

phillipjohnston, 2020-05-07:
You are on the right track. That file compiles and links successfully.

jwellbelove, 2020-05-07:
It seems it's down to how iterators are checked.
I was using things like `CHECK_EQUAL(data.begin(), std::begin(data));`
Although the member and free function should return exactly the same type, for CLang on Mac, it seems it doesn't. The unit test framework didn't know how to compare them or print the values.
I changed the style to `CHECK(data.begin() == std::begin(data));`

jwellbelove, 2020-05-09:
I'd try it out on my MacOS VM, but I've forgotten the password...

jwellbelove, 2020-05-10:
I've pushed a whole load of changes.
Let me know how well I've done.

phillipjohnston, 2020-05-11:
Will test them later today and let you know!

phillipjohnston, 2020-05-11:
The following still generate errors:

* test_deque.cpp
* test_multimap.cpp
* test_pearson.cpp
* test_string_view.cpp
* test_to_u16string.cpp
* test_to_u32_string.cpp
* test_to_string.cpp
* test_to_wstring.cpp

Only had a few minutes so I wasn't able to try to improve upon your branch, but I did capture some logs:

[deque_log.txt](https://github.com/ETLCPP/etl/files/4612404/deque_log.txt)
[multimap_log.txt](https://github.com/ETLCPP/etl/files/4612406/multimap_log.txt)
[pearson_log.txt](https://github.com/ETLCPP/etl/files/4612407/pearson_log.txt)
[string_view_log.txt](https://github.com/ETLCPP/etl/files/4612408/string_view_log.txt)
[u32string_log.txt](https://github.com/ETLCPP/etl/files/4612409/u32string_log.txt)

jwellbelove, 2020-05-12:
OK, it looks like the additional changes I did to simplify the conditional compilation macros triggered CLang to warn about undefined behaviour for a #define that contained 'defined'.
I've fixed these now.

phillipjohnston, 2020-05-15:
I've been quite swamped - sorry for the testing delay. I have some time tomorrow and will test the latest changes.

phillipjohnston, 2020-05-15:
These are still failing... seems like a reversal of fortune :)

* 'test/deque.cpp',
* 'test/test_multimap.cpp',
* 'test/test_pearson.cpp',
* 'test/test_string_view.cpp',
* 'test/test_to_u16string.cpp',
* 'test/test_to_u32string.cpp',
* 'test/test_to_string.cpp',
* 'test/test_to_wstring.cpp',
* 'test/test_type_select.cpp',
* 'test/test_unordered_map.cpp',
* 'test/test_unordered_multimap.cpp',
* 'test/test_unordered_multiset.cpp',
* 'test/test_unordered_set.cpp',

[deque.txt](https://github.com/ETLCPP/etl/files/4636897/deque.txt)
[multimap.txt](https://github.com/ETLCPP/etl/files/4636898/multimap.txt)
[test_to_u16_string.txt](https://github.com/ETLCPP/etl/files/4636899/test_to_u16_string.txt)
[test_to_u32string.txt](https://github.com/ETLCPP/etl/files/4636901/test_to_u32string.txt)
[test_pearson.txt](https://github.com/ETLCPP/etl/files/4636903/test_pearson.txt)

jwellbelove, 2020-05-15:
Damn! I thought I'd found it. I thought the log was pointing an issue with the unit test header being included first. The failing files seem to be the same ones I looked at last time. You did use the hotfix branch?

phillipjohnston, 2020-05-15:
Yes, this commit:

```
commit 23f489f7d500f70470dd825b8ee96eb6dd1ce6ed (HEAD -> hotfix/fix-unit-test-errors-for-mac-clang, origin/hotfix/fix-unit-test-errors-for-mac-clang)
Author: John Wellbelove <github@wellbelove.co.uk>
Date:   Tue May 12 10:30:51 2020 +0100

    Fixed undefined behaviour '#defines'
```

jwellbelove, 2020-05-16:
I've been trying to get a new MacOS running recently, but with limited success so far.
If I can do that then I can test it here.

finger42, 2020-07-28:
I could see similar errors with the build on Github Actions with Ubuntu 18.04 and Clang-9

The build is running on my fork:
https://github.com/finger42/etl/runs/914647293

Errors like:
```
stream << "Expected " << expected << " but was " << actual;
         ~~~~~~~~~~~~~~~~~~~~~ ^  ~~~~~~~~
/home/runner/work/etl/etl/test/test_flat_map.cpp:449:7: note: in instantiation of function template specialization 'UnitTest::CheckEqual<etl::ireference_flat_map<int, TestDataNDC<std::__cxx11::basic_string<char> >, etl::less<int> >::iterator, etl::ireference_flat_map<int, TestDataNDC<std::__cxx11::basic_string<char> >, etl::less<int> >::iterator>' requested here
      CHECK_EQUAL(data.begin(), std::begin(data));
      ^
/home/runner/work/etl/etl/test/unittest++/install/include/UnitTest++/CheckMacros.h:151:27: note: expanded from macro 'CHECK_EQUAL'
      #define CHECK_EQUAL UNITTEST_CHECK_EQUAL
                          ^
/home/runner/work/etl/etl/test/unittest++/install/include/UnitTest++/CheckMacros.h:40:17: note: expanded from macro 'UNITTEST_CHECK_EQUAL'
      UnitTest::CheckEqual(*UnitTest::CurrentTest::Results(), expected, actual, UnitTest::TestDetails(*UnitTest::CurrentTest::Details(), __LINE__)); \
                ^
/usr/bin/../lib/gcc/x86_64-linux-gnu/9/../../../../include/c++/9/ostream:245:7: note: candidate function not viable: no known conversion from 'const etl::ireference_flat_map<int, TestDataNDC<std::__cxx11::basic_string<char> >, etl::less<int> >::iterator' to 'const void *' for 1st argument; take the address of the argument with &
      operator<<(const void* __p)
```

I try to prepare a patch where 
`CHECK_EQUAL(data.begin(), std::begin(data));`
is replaced with 
`CHECK(data.begin() == std::begin(data));`

Also it is a cheap solution to test on Unix with Clang like:
https://github.com/finger42/etl/commit/c2ce089d677cd8c162dbf438fa78e374cc490941

 ```
build-clang-9-Linux:
    runs-on: ubuntu-18.04
    steps:
      - uses: actions/checkout@v2
      - name: Build with Clang 9
        run: |
          export CC=clang-9
          export CXX=clang++-9
          cmake -D BUILD_TESTS=ON ./
          make

```

finger42, 2020-07-29:
There is also a vm / image for 

`macOS Catalina 10.15 | macos-latest or macos-10.15
`

See: https://docs.github.com/en/actions/reference/workflow-syntax-for-github-actions

finger42, 2020-08-02:
I have setup a build on the macOs which runs on github:
```
AppleClang 11.0.3.11030032
Mac OS X
  10.15.6
  19G73
```
Code:
https://github.com/finger42/etl/tree/hotfix/clang-9

With build pipline:
https://github.com/finger42/etl/runs/937595466

The error msg:
```
In file included from /Applications/Xcode_11.6.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/algorithm:643:
In file included from /Applications/Xcode_11.6.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/memory:659:
/Applications/Xcode_11.6.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/iterator:733:23: error: invalid operands to binary expression ('etl::ideque<TestDataNDC<std::__1::basic_string<char> > >::iterator' and 'etl::ideque<TestDataNDC<std::__1::basic_string<char> > >::iterator')
    return __x.base() > __y.base();
           ~~~~~~~~~~ ^ ~~~~~~~~~~
/Users/runner/work/etl/etl/test/test_deque.cpp:647:19: note: in instantiation of function template specialization 'std::__1::operator<<etl::ideque<TestDataNDC<std::__1::basic_string<char> > >::iterator, etl::ideque<TestDataNDC<std::__1::basic_string<char> > >::iterator>' requested here
      CHECK(first < second);
                  ^
/Users/runner/work/etl/etl/test/unittest++/install/include/UnitTest++/CheckMacros.h:145:21: note: expanded from macro 'CHECK'
      #define CHECK UNITTEST_CHECK
                    ^
/Applications/Xcode_11.6.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/utility:603:1: note: candidate template ignored: could not match 'pair<type-parameter-0-0, type-parameter-0-1>' against 'etl::ideque<TestDataNDC<std::__1::basic_string<char> > >::iterator'
operator> (const pair<_T1,_T2>& __x, const pair<_T1,_T2>& __y)
```

I got no idea what going on here...

jwellbelove, 2020-08-03:
It _appears_ to be complaining about the '>' operator for etl::deque iterators.

jwellbelove, 2020-08-03:
One of the lines it's complaining about is in test_deque.cpp

```
    //*************************************************************************
    TEST(test_reverse_iterator_comparison)
    {
      DataNDC data(SIZE, N0);

      DataNDC::reverse_iterator first  = data.rbegin() + 1;
      DataNDC::reverse_iterator second = data.rbegin() + 4;

      CHECK(first < second);
      CHECK(!(second < first)); <<<<< THIS LINE <<<<<
    }
```
Are you compiling with or without the STL?

```
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
```
The unit tests require C++17. Could this be an issue?

finger42, 2020-08-03:
On the hotfix/clang-9 branch I let the code build with:
```
build-clang-10-osx:
  runs-on: macos-10.15
  steps:
    - uses: actions/checkout@v2
    - name: Build and run
      run: |
        export CC=clang
        export CXX=clang++
        cmake -D BUILD_TESTS=ON ./
        make
        ./test/etl_tests
```

jwellbelove, 2020-08-04:
So far I have identified two issues.
1. The xcode compiler does not like `__sync_fetch_and_add` to be used to add an integral value to a pointer. My initial experiments have been to make the internal representation of the atomic pointer template to be a `uintptr_t` with a cast back to `T*` on access.

2. xcode is complaining that `etl::deque::iterator` and `etl::deque::const_iterator` do not have comparison operators defined.
This is true and it _should_ have them due to them being defined as random access iterators.

jwellbelove, 2020-08-08:
I think I've pretty much got to the bottom of the Mac clang compilation issues now.
the hotfix branch is compiling without error, so I'm just doing to extra testing here before I push to the master branch.

jwellbelove, 2020-08-09:
@phillipjohnston Can you check that the hotfix/clang9 branch is compiling for your setup without errors?

phillipjohnston, 2020-08-14:
Everything looks good, all tests pass.

```
$ ./buildresults/etl_unit_tests
Success: 3525 tests passed.
Test time: 2.95 seconds.
```

closed 2020-10-10 by jwellbelove