ETLCPP/etl #222: ETL_NO_STL includes <new> (not existing on Arduino)

eudoxos, 2020-05-12:
I am testing ETL (latest git, not installed through the library manager) with Arduino (tools version 1.8.2, not using the IDE but Arduino-Makefile instead), having done `#define ETL_NO_STL` in `etl_profile.h`. I am getting 

    ./libraries/etl/include/etl/memory.h:45:10: fatal error: new: No such file or directory

I tried to work around this by using ArduinoSTL, but that one won't compile with c++17.

I was looking at configuration options (and the source) and did not see a way to avoid inclusion of `<new>`. Any suggestion how to move forward?  Thanks!

jwellbelove, 2020-05-12:
Did you have to implement placement new?

eudoxos, 2020-05-13:
Not at all, I just added the code above, plus an empty `new` to make the `#include<new>` not fail.

jwellbelove, 2020-05-13:
I assume Arduino must be defining that operator somewhere.
It's very frustrating that Arduino almost seems to go out of its way to be non-standard.
To fix this, I'm going to have to put in more conditional compilation to cope with Arduino's 'specialness'.

eudoxos, 2020-05-13:
I understand that frustration, they are not pushing the platform forward much in terms of compliance and up-to-date standards (perhaps not in other senses, either)...

The declaration is here https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/new.h and definition here https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/new.cpp .

I created a PR for Arduino, just in case it happens to get some upstream consideration. https://github.com/arduino/ArduinoCore-avr/pull/340

jwellbelove, 2020-05-13:
Which particular device are you using?

jwellbelove, 2020-05-13:
Do you know if the macro NAN is supported?

jwellbelove, 2020-05-13:
Cool, looks like you got the change we want on Arduino-avr

eudoxos, 2020-05-13:
I am using Arduino Nano (currently).

`NAN` is not defined, I also grepped Arduino headers, headers of avg-gcc and predefined macros via `avr-g++ -dM -E`, no `NaN` (or `nan` or `NAN`).

I am fine with workarounds, as long as there are documented and fit nicely in one place. Otherwise maintaining libs like yours would be a nightmare.

[__has_include](https://en.cppreference.com/w/cpp/preprocessor/include) is standardized since c++17, in case that helps in this case

jwellbelove, 2020-05-13:
I've just checked and #include <new> is included in 30 ETL files.

I could either add checks for Arduino or just assume that <new> will be added to Arduino-avr very shortly anyway.

jwellbelove, 2020-05-13:
I'd love to use __has_include but the ETL attempts top support everything back to C++03

grbba, 2020-06-09:
Same for me here since about two days! I really like ETL but was very close to drop it as I couldn't figure out what was happening. The workaround fixed it! Thanks a lot

grbba, 2020-06-09:
I have been too quick with my answer :( i just had all the definitions done and all that compiled properly but now starting real use I get this when calling insert to a map ... 

.pio/libdeps/megaatmega2560/Embedded Template Library/include/etl/map.h:1387:7: error: no matching function for call to 'operator new(sizetype, etl::imap<int, Locomotive, etl::less<int> >::value_type*)'

is there a setting for etl_profile ev. to force a different way to create instances in map.h ?

jwellbelove, 2020-06-09:
All of the containers use 'placement new' to create instances of objects in raw memory. There is no alternative way of doing this.

jwellbelove, 2020-06-09:
https://github.com/arduino/ArduinoCore-avr/pull/33

jwellbelove, 2020-06-09:
// declaration of placement new:
inline void *operator new(size_t, void *buf) { return buf; }

from here https://forum.arduino.cc/index.php?topic=74145.msg559001#msg559001

closed 2020-08-04 by jwellbelove