ETLCPP/etl #227: hash function for enum

firefly1250, 2020-05-24:
I'd like to use unordered_map in which key is enum.
I can use it by defining hash function by myself, but it is inconvinient.
Could you add definition of hash function for enum like std?

jwellbelove, 2020-05-24:
I looked at this a while ago, but came up against the need to implement a cross platform ETL version of `std::is_enum`. Implementing `etl::is_enum` requires several other type traits to be defined, all of which can be tricky to implement without help from the compiler. I may be able to put together a solution that will work for compilers that support the `__is_enum(T)` intrinsic.

jwellbelove, 2020-05-24:
It can be tricky trying to track down which _version_ of each compiler started to define __is_enum(T).

jwellbelove, 2022-09-12:
https://www.etlcpp.com/bitset.html

closed 2022-09-12 by jwellbelove