ETLCPP/etl #229: [suggestion] Suppress error if `etl_profile.h` not exists.

puzrin, 2020-06-02:
Latest major releases (with features autodetect) made defaults nice, and `etl_profile.h` become empty in many cases.

I suggest to make one step more - skip "#include" if `etl_profile.h` not exists. This can be detected in new compilers, with fallback to current behaviour (error) in old compilers.

https://stackoverflow.com/a/33260104/1031804.

jwellbelove, 2020-06-03:
The difficulty with this is that `etl_profile.h` is included before the headers that determine the C++ language level. It is done this way so that the user can override the automatically determined values.
For example, it may be the compiler supports C++17, but the coding rules for the project only allow features up to C++11. Having the `etl_profile.h` included first allows the user to stop` platform.h` automatically determining this.

puzrin, 2020-06-03:
I mean this way of use:

```cpp
#ifndef __has_include
#define __has_include() 1
#endif

#if __has_include("etl_profile.h")
# include "etl_profile.h"
#endif
```

This does not requires compiler detect. You still can include config first and use as before. Not sure about coding rules.

Alternative idea: option `ETL_NO_CONFIG` for compiler command line.

jwellbelove, 2020-06-03:
Thanks, I didn't realise that you could use `#ifndef __has_include`

puzrin, 2020-06-03:
While created example, i realised that `ETL_NO_CONFIG` would be ok too for my case.

https://github.com/puzrin/dispenser/blob/master/platformio.ini#L17 - i already have some external defs. Add one more is not a problem.

Disadvanage of `__has_include` is silent fail when user really needs config but it can not be found (missed include path).

jwellbelove, 2020-06-03:
etl_profile.h could possibly disappear entirely by requiring that all defines should be set in the project's configuration, whether that be the CMake file or the IDE's project settings.

puzrin, 2020-06-03:
Yes, it would be nice to have such alternate mode of ETL use [without config in file]. It's the essence of my proposal. Concrete details of implementation are not principal.

jwellbelove, 2020-09-26:
`platform.h` now includes the following lines since `18.12.7`

```
// Include the user's profile definition.
#if !defined(ETL_NO_PROFILE_HEADER) && defined(__has_include)
  #if !__has_include("etl_profile.h")
    #define ETL_NO_PROFILE_HEADER
  #endif
#endif

#if !defined(ETL_NO_PROFILE_HEADER)
  #include "etl_profile.h"
#endif
```

closed 2020-09-26 by jwellbelove