ETLCPP/etl #231: Compile error in etl/type_traits.h

AbdManian, 2020-06-09:
Get compile error with g++ 8.2

test.cpp:

```
#include "platform.h"
int main() {
	return 0;
}
```
Compiler 
```
g++ (MinGW.org GCC-8.2.0-3) 8.2.0
```
Error:
```
In file included from etl/limits.h:35,
                 from etl/platform.h:35,
                 from test_etl.cpp:1:
etl/type_traits.h:783:52: error: expected template-name before '<' token
   struct integral_constant : std::integral_constant<T, VALUE> {};
                                                    ^
etl/type_traits.h:783:52: error: expected '{' before '<' token
etl/type_traits.h:795:48: error: expected template-name before '<' token
   struct bool_constant : std::integral_constant<bool, B> { };
                                                ^
......
```

jwellbelove, 2020-06-09:
What level of C++ are you compiling to? C++03, C++11 etc

jwellbelove, 2020-06-09:
I assume you are using the STL too? i.e. ETL_NO_STL is not defined.

AbdManian, 2020-06-09:
> I assume you are using the STL too? i.e. ETL_NO_STL is not defined.

I've tested both (with/without STL defining in etl_profile.h). Also testing with -std=c++xx switches. No difference. Maybe I'm missing something

jwellbelove, 2020-06-09:
With ETL_NO_STL defined it should be using a completely different definition of `etl::integral_constant`
The one at line 80 in `type_traits.h`

AbdManian, 2020-06-09:
> With ETL_NO_STL defined it should be using a completely different definition of `etl::integral_constant`
> The one at line 80 in `type_traits.h`
Get same error even I've defined the ETL_NO_STL from command line.

```
g++ test_etl.cpp -o test_etl.exe -D__STDC_LIMIT_MACROS -DETL_NO_STL -Iinclude\etl
```

AbdManian, 2020-06-09:
Also tested with cross compiler and get same result. 
_mb-g++ (crosstool-NG 1.20.0) 7.2.0_

jwellbelove, 2020-06-09:
What version of the ETL are you using?

AbdManian, 2020-06-09:
I've tested 18.3.3 and 18.3.2

jwellbelove, 2020-06-09:
Are you saying that with ETL_NO_STL defined you still get an error at line 783 in type_traits.h?

AbdManian, 2020-06-09:
Yes
```
In file included from include\etl/limits.h:35,
                 from include\etl/platform.h:35,
                 from test_etl.cpp:1:
include\etl/type_traits.h:783:52: error: expected template-name before '<' token
   struct integral_constant : std::integral_constant<T, VALUE> {};
                                                    ^
include\etl/type_traits.h:783:52: error: expected '{' before '<' token
include\etl/type_traits.h:795:48: error: expected template-name before '<' token
   struct bool_constant : std::integral_constant<bool, B> { };
                                                ^
include\etl/type_traits.h:795:48: error: expected '{' before '<' token
include\etl/type_traits.h:814:72: error: expected template-name before '<' token
   template <typename T> struct remove_reference : std::remove_reference<T> {};
                                                                        ^
include\etl/type_traits.h:814:72: error: expected '{' before '<' token
include\etl/type_traits.h:824:68: error: expected template-name before '<' token
   template <typename T> struct remove_pointer : std::remove_pointer<T> {};
                                                                    ^
include\etl/type_traits.h:824:68: error: expected '{' before '<' token
include\etl/type_traits.h:834:62: error: expected template-name before '<' token
   template <typename T> struct add_pointer : std::add_pointer<T> {};
                                                              ^
include\etl/type_traits.h:834:62: error: expected '{' before '<' token
include\etl/type_traits.h:844:56: error: expected template-name before '<' token
   template <typename T> struct is_const : std::is_const<T> {};
                                                        ^
include\etl/type_traits.h:844:56: error: expected '{' before '<' token
include\etl/type_traits.h:854:64: error: expected template-name before '<' token
   template <typename T> struct remove_const : std::remove_const<T> {};
                                                                ^
include\etl/type_traits.h:854:64: error: expected '{' before '<' token
include\etl/type_traits.h:864:58: error: expected template-name before '<' token
   template <typename T> struct add_const : std::add_const<T> {};
                                                          ^
include\etl/type_traits.h:864:58: error: expected '{' before '<' token
include\etl/type_traits.h:874:62: error: expected template-name before '<' token
   template <typename T> struct is_volatile : std::is_volatile<T> {};
                                                              ^
include\etl/type_traits.h:874:62: error: expected '{' before '<' token
include\etl/type_traits.h:884:70: error: expected template-name before '<' token
   template <typename T> struct remove_volatile : std::remove_volatile<T> {};
                                                                      ^
include\etl/type_traits.h:884:70: error: expected '{' before '<' token
include\etl/type_traits.h:894:64: error: expected template-name before '<' token
   template <typename T> struct add_volatile : std::add_volatile<T> {};
                                                                ^
include\etl/type_traits.h:894:64: error: expected '{' before '<' token
include\etl/type_traits.h:904:58: error: expected template-name before '<' token
   template <typename T> struct remove_cv : std::remove_cv<T> {};
                                                          ^
include\etl/type_traits.h:904:58: error: expected '{' before '<' token
include\etl/type_traits.h:914:52: error: expected template-name before '<' token
   template <typename T> struct add_cv : std::add_cv<T> {};
                                                    ^
include\etl/type_traits.h:914:52: error: expected '{' before '<' token
include\etl/type_traits.h:924:62: error: expected template-name before '<' token
   template <typename T> struct is_integral : std::is_integral<T> {};
                                                              ^
include\etl/type_traits.h:924:62: error: expected '{' before '<' token
include\etl/type_traits.h:934:58: error: expected template-name before '<' token
   template <typename T> struct is_signed : std::is_signed<T> {};
                                                          ^
include\etl/type_traits.h:934:58: error: expected '{' before '<' token
include\etl/type_traits.h:944:62: error: expected template-name before '<' token
   template <typename T> struct is_unsigned : std::is_unsigned<T> {};
                                                              ^
include\etl/type_traits.h:944:62: error: expected '{' before '<' token
include\etl/type_traits.h:954:74: error: expected template-name before '<' token
[... 314 lines cut ...]
In file included from include\etl/container.h:37,
                 from include\etl/algorithm.h:44,
                 from include\etl/char_traits.h:37,
                 from include\etl/limits.h:36,
                 from include\etl/platform.h:35,
                 from test_etl.cpp:1:
include\etl/iterator.h:279:5: error: 'ETL_CONSTEXPR17' does not name a type
     ETL_CONSTEXPR17 reverse_iterator& operator=(const reverse_iterator<TOther>& other)
     ^~~~~~~~~~~~~~~
include\etl/iterator.h:279:5: note: the macro 'ETL_CONSTEXPR17' had not yet been defined
In file included from test_etl.cpp:1:
include\etl/platform.h:152: note: it was later defined here
 #define ETL_CONSTEXPR17
 
In file included from include\etl/container.h:37,
                 from include\etl/algorithm.h:44,
                 from include\etl/char_traits.h:37,
                 from include\etl/limits.h:36,
                 from include\etl/platform.h:35,
                 from test_etl.cpp:1:
include\etl/iterator.h:286:5: error: 'ETL_CONSTEXPR17' does not name a type
     ETL_CONSTEXPR17 TIterator base() const
     ^~~~~~~~~~~~~~~

```

jwellbelove, 2020-06-09:
I don't quite know how that is happening, as when ETL_NO_STL is defined, that whole section of code is disabled by the pre-processor. Do you have an editor that will show you what is enabled/disabled?

jwellbelove, 2020-06-09:
I compile the unit tests using GCC 8.4.0 as well as other compilers.

AbdManian, 2020-06-09:
> I don't quite know how that is happening, as when ETL_NO_STL is defined, that whole section of code is disabled by the pre-processor. Do you have an editor that will show you what is enabled/disabled?

I've create preprocessor output with "-E" and "-g3" switches. It may be useful.
[test_etl.txt](https://github.com/ETLCPP/etl/files/4752608/test_etl.txt)

AbdManian, 2020-06-09:
> I compile the unit tests using GCC 8.4.0 as well as other compilers.

I have GCC 8.2.0. Maybe there is issue with this version or I'm missing something in using/setup etl.

jwellbelove, 2020-06-09:
I've just tried your test.cpp with your minimal 'main()' + an empty etl_profile.h with GCC 8.1 on Windows.
I used the command line 
g++ test.cpp -o test.exe -D__STDC_LIMIT_MACROS -DETL_NO_STL -Iinclude\etl
Everything compiled without error.

So did
g++ test.cpp -o test.exe -D__STDC_LIMIT_MACROS -Iinclude\etl

AbdManian, 2020-06-09:
Same result in linux _GCC (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0_
I think I am missing something.

AbdManian, 2020-06-09:
I found something. If _platform.h_ is included as _etl/platform.h_ compile will be successful.

Successful:
```c++
// test1.cpp
// g++ test1.cpp -o test1 -DETL_NO_STL -Iinclude

#include "etl/platform.h"
int main() {
    return 0;
}
```

Compile error:
```c++
// test2.cpp
// g++ test2.cpp -o test2 -DETL_NO_STL -Iinclude/etl

#include "platform.h"
int main() {
    return 0;
}
```

jwellbelove, 2020-06-09:
Ah yes, that's the preferred include path setup. 
Maybe there was a header name clash somewhere between the default include search paths and the ETL and the incorrect one was chosen.

AbdManian, 2020-06-09:
Yes. Probably incorrect included file causes the issue.
ETL is a great work. Thank you.

closed 2020-08-03 by jwellbelove