ETLCPP/etl #232: question about const correctness in pool.h

jeremyerdmann, 2020-06-15:
I'm just getting back into trying the ETL at our company.  It's neat to see all the recent progress!  When I compile the code with IAR Embedded Workbench's MISRAC 2004 checkers enabled (ver. 8.42.2) I stumbled across the following two warnings that I though I should probably look into (I get others too, but they seem pretty benign).  I also paused on the commented out template parameter on line 277. 

```
Error[Pm139]: conversions shall not be performed between a pointer to object and any type other than an integral type, \include\etl\pool.h:259 another pointer to object type or a pointer to void. This converts from type "void const *" to "char *" (MISRA C 2004 rule 11.2) 
Error[Pm139]: conversions shall not be performed between a pointer to object and any type other than an integral type,  \include\etl\pool.h:280 another pointer to object type or a pointer to void. This converts from type "void const *" to "char const *" (MISRA C 2004 rule 11.2) 
```

Not surprisingly, changing the casts on lines 259 and 280 to C++ static_casts complains about invalid type conversions:

```
Error[Pa073]: invalid type conversion (from "void const *const" to "char *") \include\etl\pool.h:259 
Error[Pa073]: invalid type conversion (from "void const *" to "char *") \include\etl\pool.h:280 
```

I think I'm OK with `is_item_in_pool()` needing a reinterpret_cast to go from a const void* (or perhaps it should be const T*) to a const char*.  However, I think it probably can't be in the release methods because of the freeing process messing with pointers and for sure not in the destroy methods where we call the destructor on T.

After some tinkering with the code, I ended up with the attached file.
[pool.txt](https://github.com/ETLCPP/etl/files/4783240/pool.txt)

I build in IAR and have a few unit tests of my use of the pool.  I had started getting the build environment setup so I could possibly submit as a PR, but I ran into a couple of snags and haven't had enough free time to investigate them.  I see someone created a slack conversation on the topic, perhaps I'll find the information I need to get unstuck.  If I do, I'll consider issuing a PR with this change provided tests all pass.

Lastly, it seemed like an opportunity for a shameless plug for East const :): https://mariusbancila.ro/blog/2018/11/23/join-the-east-const-revolution/.

jwellbelove, 2020-08-03:
Unfortunately changing to `void release(void* p_object)` causes the calls to release to not compile in `void destroy(const U* const p_object)`

Not saying that there _may_ be a permutation that does not show warnings.

jwellbelove, 2020-09-26:
Just had a look at this again.
Do these help?

```
    void release(const void* const p_object)
    {
      const uintptr_t p = uintptr_t(p_object);
      release_item((char*)p);
    }
```

```
    bool is_in_pool(const void* p_object) const
    {
      const uintptr_t p = uintptr_t(p_object);
      return is_item_in_pool((const char*)p);
    }
```

jeremyerdmann, 2020-10-17:
Thanks for the bump.  It got lost with all the other updates.  I'll take a
look on Monday and get back to you.

Jeremy

jeremyerdmann, 2020-10-22:
Sorry again for the delay in responding.  I hadn't looked at that in so long, so I had to get my bearings again (June 15th was the initial post... I struggle to remember what I did on Friday, come Monday).  The short answer is yes, it suppresses the warning.  

However, it feels like cheating casting to a type that will hold any type of pointer as that feel like were just doing a const_cast in a round about way.  I prefer the C++ static_cast<>(...) syntax over C-style casts as it forces me to be explicit for all other cast types (i.e. did you really want to do a reinterprest_cast<> or the somewhat evil const_cast<>?).  To be fair, MISRA-C is not a great checker for C++ code, and these 2 were only a small portion of the checks that failed for me... this one just stuck out.  In the end, it's probably not a problem, but it could be suppressed.

I went back to the initial file I had attached (attached is a quick diff to clarify what I was getting at back in June).  Besides tweaking these methods, I also needed to remove the const in front of T in destroy(...)'s signature shouldn't be there (still compiles w/ 0 warnings/errors).  My thinking was the pointer needs to be const, but the data is technically being modified, so it cannot.  I'm probably missing something, but I found if you remove that const from destroy, I thought it would allow you to make the other changes.  Looks like I also changed from a void* to a const* in another place.   Maybe it's worth trying again, but like I said, it's probably fine as-is too (didn't stop me from using it).  Just a heads up that I'm still running the older version ETL from in June, so it's possible some other changes have happened since then also.

As an aside, we're moving away from MISRA, in favor of Clang-Tidy.  Not sure if you played with it yet, but it has lots and lots of checks, and in our case it gets rid of the dumb false positives from normal C++ (e.g. the extra braces to place something in a namespace).

[image]

closed 2020-11-27 by jwellbelove