ETLCPP/etl #236: Feature request: bitfield::to_ulong conversion

positron96, 2020-06-24:
Hi. 
Would it be possible to implement bitfield::to_ulong function for places where bitfield needs to be e.g. passed as bytes to other devices. As I understand, there is a corresponding function in STL.

jwellbelove, 2020-06-24:
`etl::bitset` does not have a `to_ulong()` but it does have a more useful `value<T>()` template function.

The equivalent of `to_ulong()` would be `value<unsigned long>()`

positron96, 2020-06-24:
Oh, thank you very much then, problem solved!

closed 2020-06-24 by positron96

positron96, 2020-06-27:
Also, somewhat related: can I set multiple bits at once?

jwellbelove, 2020-06-27:
The only way to do that is to &|^ with another bitset.

positron96, 2020-06-28:
Again, thank you! For the moment, I'm recreating a bitset from a modified uint32_t.