ETLCPP/etl #237: Intrusive list: is_linked() after unlink

AndWass, 2020-07-03:
Using an intrusive list with bidirectional iterators, `is_linked()` returns true after `unlink()`. Is this by design? And if so, any chance this can be changed?

jwellbelove, 2020-07-03:
That is a mistake. unlink() should clear the next/previous pointers of etl::bidirectional_link

AndWass, 2020-07-03:
Thanks for very swift response! I fixed it locally, but not ready to do a pull-request yet though

jwellbelove, 2020-07-03:
I had to modify etl::intrusive_list::splice() too, as clearing the links in unlink stopped a range being spliced into another etl::intrusive_list

AndWass, 2020-07-03:
Oh ok, didn't notice that. Will leave you to it instead :) Many thanks for swift responses, much appreciated!

jwellbelove, 2020-07-04:
How important is it that the links are cleared for you? The more I've investigated the issue, the more I regret adding is_linked() to the API.
To ensure that links are cleared whenever an intrusive object is erased from an intrusive container, the list of erased objects must be traversed. This would severely impact the performance of the intrusive containers, as currently, erasing a range merely requires the modification of four pointers (for bidirectional links). To clear them would either require the unlink function or container to walk the erased section of the list, clearing the link pointers. This could severely impact users who are currently using the intrusive containers for their high speed.

AndWass, 2020-07-04:
`is_linked()` isn't actually important (and it sounds like maybe it should be deprecated and removed in a later release maybe, since as you say it can lie at times anyway?).

The main issue as it stands now is that it's not safe to call `unlink()` multiple times unless you call clear after every `unlink()`. It is also not safe to call `unlink()` on an item that has previously been erased from the list.

```cpp
#include "etl/intrusive_list.h"
#include "etl/intrusive_links.h"
#include <cassert>

struct test: etl::bidirectional_link<0>
{
    int value = 0;
};

int main() {
    etl::intrusive_list<test> list;
    test item1;
    item1.value = 1;

    test item2;
    item2.value = 2;

    list.push_back(item1);
    item1.unlink();
    //list.erase(list.begin()); // This will cause the assert below to fail as well
    list.push_back(item2);
    item1.unlink();
    assert(list.begin() != list.end()); // This assert will fail
}
```

Would it be possible clear the links after an unlink and single item erase (havent checked range erase) and deprecate (or remove) `is_linked()`?

jwellbelove, 2020-07-04:
The link/unlink API was only added for easier management of objects within an ETL or custom user intrusive container.
I didn't really see them being used outside of that.

```
    list.push_back(item1);
    list.erase(list.begin());

    list2.push_back(item1); // No manual unlink() or clear() required.
    list2.erase(list2.begin());

    item1.clear();
```
I thought about just clearing for single erases, but I don't like the idea of such an inconsistency in the API.
I did think about somehow making the, properly terminated, unlinked range available in some way for the user to walk if required.

jwellbelove, 2020-07-04:
There is also a difficulty in efficiently accessing the last object in an erase range if the objects are forward_links. The container's erase range API requires the one before the first, to the one after the last. This makes clearing the etl_next pointer on the last object impossible without walking the entire unlinked range.

AndWass, 2020-07-04:
Ok, I'm all out of ideas then. Also unlinking anywhere basicly makes `size()` unusable, so it's probably best to keep it internal. Maybe rename it to `internal_unlink()` or something to make i clearer that it is meant to be an internal function? Other than that I have no more suggestions that wouldn't require a major redesign (and increased complexity).

jwellbelove, 2020-07-04:
I should probably make it clear on the website that the link/unlink API is intended for the internal management of ETL or user containers and that you should not use both together.

GatorQue, 2020-07-04:
Does it make sense to use a "friend" class designation instead? Is that
even possible with templates?

closed 2020-08-03 by jwellbelove