ETLCPP/etl #238: etl/span.hpp wrong use of constexpr for c++11

timrid, 2020-07-10:
In the following enviornment:
- etl-18.4.0
- Keil uVision
- armclang v6 Compiler
- c++11 enabled

when i write:

```
uint8_t array[16];
etl::span<uint8_t> testSpan1(array);
etl::span<uint8_t> testSpan2 = testSpan1;
```

i get the following warnings:

```
../library/etl-18.4.0/include\etl/span.h(237): warning: 'constexpr' non-static member function will not be implicitly 'const' in C++14; add 'const' to avoid a change in behavior [-Wconstexpr-not-const]
    ETL_CONSTEXPR span& operator =(const span& other) ETL_NOEXCEPT
                        ^
                                                      const
../library/etl-18.4.0/include\etl/span.h(239): warning: use of this statement in a constexpr function is a C++14 extension [-Wc++14-extensions]
      mbegin = other.mbegin;
      ^
../library/etl-18.4.0/include\etl/span.h(299): warning: use of this statement in a constexpr function is a C++14 extension [-Wc++14-extensions]
      if (count == etl::dynamic_extent)
      ^
../library/etl-18.4.0/include\etl/span.h(303): warning: multiple return statements in constexpr function is a C++14 extension [-Wc++14-extensions]
      return etl::span<element_type, etl::dynamic_extent>(mbegin + offset, mbegin + offset + count);
      ^
../library/etl-18.4.0/include\etl/span.h(301): note: previous return statement is here
        return etl::span<element_type, etl::dynamic_extent>(mbegin + offset, mend);
        ^
```
To fix this i have to write:
`ETL_CONSTEXPR14 span& operator =(const span& other) ETL_NOEXCEPT`

And a similar warning:

```
../library/etl-18.4.0/include\etl/span.h(299): warning: use of this statement in a constexpr function is a C++14 extension [-Wc++14-extensions]
      if (count == etl::dynamic_extent)
      ^
../library/etl-18.4.0/include\etl/span.h(303): warning: multiple return statements in constexpr function is a C++14 extension [-Wc++14-extensions]
      return etl::span<element_type, etl::dynamic_extent>(mbegin + offset, mbegin + offset + count);
      ^
../library/etl-18.4.0/include\etl/span.h(301): note: previous return statement is here
        return etl::span<element_type, etl::dynamic_extent>(mbegin + offset, mend);
        ^
```

To fix this i have to write:
`ETL_CONSTEXPR14 etl::span<element_type, etl::dynamic_extent> subspan(size_t offset, size_t count = etl::dynamic_extent) const`

Is this the right way to fix it?

jwellbelove, 2020-07-10:
Yes, that is the correct way to fix it. Can you do a pull request?

timrid, 2020-07-20:
Thanks for fixing this bug so fast. But you only corrected the one spot. In the assignment operator the problem still exists.

I tried to make a pull-request, but unfortunately I do not have the permissions to create a new branch for the pull-request.

jwellbelove, 2020-07-20:
Fast! I thought I'd been a bit slow in not following it up until now 😄

jwellbelove, 2020-07-21:
https://opensource.com/article/19/7/create-pull-request-github

jwellbelove, 2020-07-21:
Or can you attach the modified file to a message here.

closed 2020-08-02 by jwellbelove