ETLCPP/etl #242: Flat_map: TMapped& at(key_parameter_t key) returns wrong element if key is not in table

eliasvoe, 2020-07-20:
The TMapped& at(key_parameter_t key) method in the flat_map class returns a wrong element if it is called with a key that has not been added to the flat_map before instead of throwing the etl::flat_map_out_of_range exception.
If I use the exact same could but change the etl::flat_map to a etl::map it correctly throws the etl::map_out_of_bounds exception.

jwellbelove, 2020-07-20:
In the unit tests, `etl::flat_map` throws an `etl::flat_map_out_of_bounds` exception on a non-existent key.

See `test_flat_map.cpp` test at line 575

eliasvoe, 2020-07-20:
I have done further testing and identified that the exception is only thrown when no greater key exists in the etl::flat_map. But when there are greater keys in the map, the element with the next greater key is returned by the at() method.

jwellbelove, 2020-07-20:
OK, thanks. I'll try to get that that fixed later today.

jwellbelove, 2020-07-20:
Fixed v18.8.2

closed 2020-07-20 by jwellbelove