ETLCPP/etl #243: move of etl::vector with external buffer

timrid, 2020-07-20:
In the following enviornment:

etl-18.4.0
Keil uVision
armclang v6 Compiler
c++11 enabled

I use the vector with an external buffer as in the following example:

```
uint8_t data[100];
etl::vector<uint8_t, 0> vect( nullptr, 0 );
vect = etl::vector<uint8_t, 0>( &data[0], 100 );

int sz = vect.max_size();
```

I would assume that `vect.max_size()` now returns me 100. However, it is still 0, so the size is not copied in the move assignment operator.

Am I wrong with my assumption, or is this a bug?

jwellbelove, 2020-07-20:
I'll try to look at that later today.

jwellbelove, 2020-07-21:
With the current ETL code, you need to specify the max size in the `vect `constructor.

uint8_t data[100];
etl::vector<uint8_t, 0> vect( nullptr, **100** );
vect = etl::vector<uint8_t, 0>( &data[0], 100 );

int sz = vect.max_size();

timrid, 2020-07-22:
I agree with you that for the current implementation this must be done as you wrote in the example.

However, this behavior is not what you would expect with an std::move of a vector with an external buffer. According to https://en.cppreference.com/w/cpp/language/move_assignment the move assignment operator "typically steal the resources held by the argument".

I rewrote the example a bit, added the results of the etl-18.4.0 and what i would expect:

```
uint8_t data[100];
etl::vector<uint8_t, 0> vect1( &data[0], 100 );
etl::vector<uint8_t, 0> vect2( nullptr, 0U );
vect1.push_back(42);
vect1.push_back(43);

// give all resources from vect1 to vect2
vect2 = std::move(vect1);

// vect1 should have no external buffer any more
uint8_t* beg1 = vect1.begin();  // beg1=&data[0]    (expected beg2=nullptr)
uint8_t* end1 = vect1.end();    // beg2=&data[0]    (expected end2=nullptr)
int cap1 = vect1.capacity();    // cap1=100         (expected sz2=0)
int siz1 = vect1.size();        // siz=0            (as expected)
    
// vect2 should have full control over the external buffer "data[100]"
uint8_t* beg2 = vect2.begin();  // beg2=&data[0]    (as expected)
uint8_t* end2 = vect2.end();    // end2=&data[2]    (as expected)
int cap2 = vect2.capacity();    // cap2=0           (expected cap2=100)
int siz2 = vect2.size();        // siz2=2           (as expected)
```

In the results of the etl-18.4.0 the vect1 and vect2 are now mixed. A part of the infos about the buffer are in vect1, another part is in vect2.

**Solution:**
My solution to this problem would be as follows:

1. `etl::vector_base::CAPACITY` must be non-const
2. change move assignment operator of `etl::vector<T, 0>` to this:
```
vector& operator = (vector&& rhs)
{
  if (&rhs != this)
  {
    this->clear();
    this->p_buffer = rhs.p_buffer;
    this->p_end    = rhs.p_end;
    this->CAPACITY = rhs.CAPACITY;
    rhs.p_buffer = nullptr;
    rhs.p_end = nullptr;
    rhs.CAPACITY = 0U;

    this->initialise_destination_external_buffer_after_move();
    rhs.initialise_source_external_buffer_after_move();
  }

  return *this;
}
```
3. change move constructor of `etl::vector<T, 0>` to this:
```
vector(vector&& other)
  : etl::ivector<T>(other.p_buffer, other.CAPACITY)
{
  if (this != &other)
  {
    this->p_end    = other.p_end;
    other.p_buffer = nullptr;
    other.p_end = nullptr;
    other.CAPACITY = 0U;

    this->initialise_destination_external_buffer_after_move();
    other.initialise_source_external_buffer_after_move();
  }
}
```

jwellbelove, 2020-07-22:
Yes, I agree, I took a quick look yesterday and that's how I was planning to change the code.

jwellbelove, 2020-07-22:
Further work on the code has shown up an unresolvable issue with moving a vector that has an external buffer.
The sticking point is that the `ivector` move assignment operator cannot be made to work with this type of vector.
There is no way to stop this from happening.

```
uint8_t data[100];
etl::vector<uint8_t, 100> vect1;
etl::vector<uint8_t, 0> vect2( nullptr, 0U );
vect1.push_back(42);
vect1.push_back(43);

etl::ivector<uint8_t>& ivect1 = vect1;
etl::ivector<uint8_t>& ivect2 = vect2;

// OOPS! vect1 is now broken.
vect2 = std::move(vect1);
```
It looks like the best course is not to allow move operations for externally buffered vectors (or create a custom `move` member function).

There appears to be a similar issue with moving `etl::list` & `etl::forward_list` using shared pools.

jwellbelove, 2020-07-22:
I'm tempted to keep the current code and make it mandatory to assign a non-null buffer.

jwellbelove, 2020-07-23:
Yet another option is to add a boolean flag to `etl::vector` to indicate whether it has a shared buffer or not and switch on the four permutations of internal/external buffer.

timrid, 2020-07-23:
I have also thought about the topic again and think you have to distinguish completely between:
- owning containers: Completely responsible for its own storage.
- non-owning containers: The memory is provided from outside and it saves only pointers to this memory

The current implementation of the `etl::vector<T, 0>` is kind of a mixture of both, because with a copy assignment it creates a copy of each element, and with a std::move it only copys the pointers. The `etl::vector<T, 42>` is a pure owning container, because it copys all elements in the copy assignment and moves all elements in the move assignment. This is a not quite straightforward behaviour within the `etl::vector<T, 0>`.

I would suggest to implement the `etl::vector` completely as an owning container, i.e. to copy the move methods from the `etl::vector<T, 42>` to the `etl::vector<T, 0>`. This would then necessarily require that it is mandatory to assign a non-null buffer, as in your idea.

In addition, you can create a new completely non-owning vector container type, which works only with external buffers and does not allocate its own memory. I see this very similar to an `etl::span`. In this case a copy assignment as well as a move assignment (if there is one at all) would only copy the pointers to the external buffer, but not copy/move of all elements. So a name like something `etl::span_vector` would make sense.

closed 2020-08-02 by jwellbelove