ETLCPP/etl #244: etl::observer calling remove_observer in notification skips the next observer from being notified

samschiedermayer, 2020-07-24:
This is the environment being used:
etl-18.3.3
iar-8.42.2
c++11 enabled

I have an observable that has three observers that are registered to it. Each of these observers are set up to unsubscribe from the observable after a certain number of notifications being received. The issue that arises is upon calling `observable.remove_observer(*this)`  from within the notification function in the observer. Since the vector that is storing the observers is modified in the process of being iterated through, the next observer gets skipped, while the last observer gets called an extra time. The expected/desired behavior would be that nothing else is affected by the removal.

Attached, I have a cpp file that shows this behavior, as well as output txt files with expected vs actual output.
[ExampleForIssue.cpp.txt](https://github.com/ETLCPP/etl/files/4973020/ExampleForIssue.cpp.txt)
[actual_output.txt](https://github.com/ETLCPP/etl/files/4973019/actual_output.txt)
[expected_output.txt](https://github.com/ETLCPP/etl/files/4973021/expected_output.txt)

A proposed solution for this issue is to iterate through the observers to notify *in a backwards fashion* in the `notify_observers` function call.

This is the code as it exists today:
```cpp
#if ETL_CPP11_SUPPORTED && !defined(ETL_OBSERVER_FORCE_CPP03)
    template <typename... TNotification>
    void notify_observers(TNotification... n)
    {
      for (auto observer : observer_list)
      {
        observer->notification(n...);
      }
    }
#else
    template <typename TNotification>
    void notify_observers(TNotification n)
    {
      for (size_t i = 0; i < observer_list.size(); ++i)
      {
        observer_list[i]->notification(n);
      }
    }
#endif
```
This is the proposed change (or something of this nature):
```cpp
// I could not find a clean way with pure c++11 to do this
template <typename TNotification>
void notify_observers(TNotification n)
{
  for (int32_t i = observer_list.size() - 1; i >= 0; --i)
  {
    observer_list[i]->notification(n);
  }
}
```

This seems to be a common enough use case for an observer/observable pattern that I think it would make sense for this functionality to work. There are two potential issues with this fix, which may or may not be important. One is the fact that if observer A removes observer B from the list of observers the issue will not be fixed, and it will only be fixed for when observer A or observer B removes itself from the observer list. The second potential issue comes into play if the order that the observers are notified matters, as if it does they will be notified in a different order. I do not believe that either of these are large causes for concern, but I may be wrong.

If it is not desired to have this fixed in the etl, having documentation about how the observer/observable module behaves in this scenario would be a good idea.

jwellbelove, 2020-07-24:
Thanks, I'll take a look tomorrow.

jwellbelove, 2020-07-25:
Ideally, 'notification()' would set a flag to say that it was running, and just mark the observer for removal after the loop had finished, but there is no way to implement this without using memory barriers or atomic flags to stop the compiler from optimising it away. As not all platforms have atomics I can't use this as a universal solution.

Another solution would be to have the ability to erase observers in a notification only if the platform supports atomics.
I'm not keen on this solution as it makes the effect of `remove_observer` inconsistent between platforms.

One solution could be to have a separate `mark_observer_for_removal` that would set a flag for the observer to be ignored whilst iterating the list and removed before returning from `notify_observers`. 

Could it be useful to have a enable/disable API for the observers, so that they can be disabled without removing from the list?

jwellbelove, 2020-07-25:
As for the order of notifications, I deliberately omitted any mention in the documentation of making any guarantees so as not to limit the implementation options.

jwellbelove, 2020-07-25:
Re: atomic flags. I can't convince myself one way or the other that the 'in_notify_observers' flag would be optimised away of not, if another member function references it.

jwellbelove, 2020-07-25:
After some thought and experimentation, I'm more inclined to add `enable`/`disable` methods that can be called from the overridden `notification` functions, and document that `remove_observer` must not be called from `notification`.

jwellbelove, 2020-07-25:
I think the best option for erasing observers after calling `notify_observers` is to add a flag to your own derived observer classes and clear them after the call.

closed 2020-07-31 by jwellbelove