ETLCPP/etl #248: clang-9: warning: array index -1

finger42, 2020-07-27:
During the setup to compile the tests with clang-9 I get some warnings of this type:

[ 12%] Building CXX object CMakeFiles/etl_tests.dir/test_multimap.cpp.o
/home/user/etl/test/test_array_wrapper.cpp:264:20: warning: array index -1 is before the beginning of the array [-Warray-bounds]
      CHECK_EQUAL(&data5[Data5::REND], &*aw5.rend());
                   ^     ~~~~~~~~~~~

Data5::REND is set to -1 , how could the warning be solved?

jwellbelove, 2020-07-27:
The REND index is like the iterator returned from an 'rend()' member function; It's a 'one-past-the-end' index. It is safe because this element is never intended to be accessed, only its index is compared.

finger42, 2020-07-28:
Yes, I understand this. But the compiler warns me to do this.

data5[Data5::REND] is like data5[-1]

Whats about to not use the -1 in the array and instead do something like:

TEST(test_dereference_rend_should_give_value_of_first_element)
{
   Data5 aw5;
   CHECK_EQUAL(0, *aw5.rend() );
}

jwellbelove, 2020-07-28:
`CHECK_EQUAL(0, *aw5.rend() );`

That is undefined behaviour. You are dereferencing an element outside the array wrapper.

jwellbelove, 2020-07-28:
`rend()` returns an iterator to the element before the start of the array.

finger42, 2020-07-28:
> `rend()` returns an iterator to the element before the start of the array.
Oh yes. My fault. Sry for that, I should read the docu first.
> 
> `data5.rend()` returns a pointer to `data5[-1]`.
> 
> You could maybe fool the compiler by using `&data5[0] - 1`

I found that there is a "fix" for the warning in Line 33:

#if defined(ETL_COMPILER_CLANG)
  #pragma clang diagnostic push
  #pragma clang diagnostic ignored "-Warray-bounds"
#endif

With this, the warning should be disabled. But not in my case. I will dig into this....

finger42, 2020-07-28:
Ok, if ETL_COMPILER_CLANG is set in the etl_profie, the warnings are gone for clang.

closed 2020-07-28 by finger42

jwellbelove, 2020-07-28:
OK, there must be something wrong in `determine_compiler.h`

clang is checked by this

```
  #if !defined(ETL_COMPILER_TYPE_DETECTED) && !defined(ETL_COMPILER_CLANG)
    #if defined(__clang__) || defined(__llvm__)
      #define ETL_COMPILER_CLANG
      #define ETL_COMPILER_TYPE_DETECTED
    #endif
  #endif
```

Is it being caught by an earlier test on your system?