ETLCPP/etl #254: Using clang-format and .clang-format as style file

finger42, 2020-08-01:
To enable clean codestyle there should be an .clang-format file.

With this we could format from IDEs and CI the code.

finger42, 2020-12-17:
How could I help to bring this mainline? 
Which type of codestyle should be used?

There are some basic styles. Or should we use our etl specific style ?

https://clang.llvm.org/docs/ClangFormatStyleOptions.html

BasedOnStyle (string)
LLVM A style complying with the LLVM coding standards
Google A style complying with Google’s C++ style guide
Chromium A style complying with Chromium’s style guide
Mozilla A style complying with Mozilla’s style guide
WebKit A style complying with WebKit’s style guide
Microsoft A style complying with Microsoft’s style guide
GNU A style complying with the GNU coding standards

daniel-brosche, 2021-10-02:
initial version proposed in scope of #436

jwellbelove, 2021-10-07:
Now added