ETLCPP/etl #259: Errors in etl::pool's create function

phillipjohnston, 2020-08-14:
I updated an older project to use the latest ETL, and pool code doesn't compile due to candidate mismatches. Is there a change in usage?

```
../src/os/freertos/os.cpp: In static member function 'static embvm::VirtualEventFlag* os::freertos::freertosOSFactory_impl::createEventFlag_impl()':
../src/os/freertos/os.cpp:84:42: error: no matching function for call to 'etl::pool<os::freertos::EventFlag, 4>::create<os::freertos::EventFlag>()'
   84 |  return event_factory_.create<EventFlag>();
      |                                          ^
In file included from ../src/os/freertos/os.cpp:4:
../subprojects/etl/include/etl/pool.h:709:8: note: candidate: 'T* etl::pool<T, SIZE_>::create(Args&& ...) [with Args = {os::freertos::EventFlag}; T = os::freertos::EventFlag; unsigned int SIZE_ = 4]'
  709 |     T* create(Args&&... args)
      |        ^~~~~~
../subprojects/etl/include/etl/pool.h:709:8: note:   candidate expects 1 argument, 0 provided
```

I also see:

```
../src/os/freertos/os.cpp: In static member function 'static embvm::VirtualMutex* os::freertos::freertosOSFactory_impl::createMutex_impl(embvm::mutex::type, embvm::mutex::mode)':
../src/os/freertos/os.cpp:71:38: error: cannot convert 'embvm::mutex::type' to 'os::freertos::Mutex&&'
   71 |  return mutex_factory_.create<Mutex>(type, mode);
      |                                      ^~~~
      |                                      |
      |                                      embvm::mutex::type
In file included from ../src/os/freertos/os.cpp:4:
../subprojects/etl/include/etl/pool.h:722:21: note:   initializing argument 1 of 'T* etl::pool<T, SIZE_>::create(Args&& ...) [with Args = {os::freertos::Mutex, embvm::mutex::mode&}; T = os::freertos::Mutex; unsigned int SIZE_ = 4]'
  722 |     T* create(Args&&... args)
      |               ~~~~~~^~~~~~~~
```

jwellbelove, 2020-08-14:
Could it have been this change?
[https://github.com/ETLCPP/etl/releases/tag/18.3.2](https://github.com/ETLCPP/etl/releases/tag/18.3.2)

phillipjohnston, 2020-08-14:
No, but that was helpful. I found 18.0.0 which was a breaking API change. My problem was that I still had code doing this:

```
return cv_factory_.create<ConditionVariable>();
```

Removing the template type makes everything work. And I like the new interface much better.

closed 2020-08-14 by phillipjohnston