ETLCPP/etl #264: Specify mutability of array_view by template parameter instead of define

bku-sue, 2020-09-14:
Currently the array_view mutability is controlled by `ETL_ARRAY_VIEW_IS_MUTABLE` symbol. I'd prefer if it could be controlled by a second template parameter, e.g. like this:
```C++

  template <typename T, bool IsMutable
#if defined(ETL_ARRAY_VIEW_IS_MUTABLE)
 = true
#else
 = false
#endif
>
  class array_view
  {
  };
```
I'm not very well versed in template programming, but my understanding is that this should be doable with `etl::enable_if<>` semantics.

jwellbelove, 2020-09-14:
Thanks for the idea, I'll take a look.

jwellbelove, 2020-09-26:
It would require a slightly more complex change, as it would involve a specialisation of the array_view template.

bku-sue, 2020-09-28:
I'm open to any alternative solution that allows one to have both mutable and non-mutable array_views in the same build.

jwellbelove, 2020-09-28:
The original reason I added the macro was that I first made the array_view mutable. I then changed it to immutable to match the STL, but used a macro to enable the old functionality.

jwellbelove, 2020-09-28:
Have you looked at `etl::span` as an alternative?

bku-sue, 2020-09-28:
I wasn't aware of it yet, it seems like what I'm looking for indeed. I just opened #277 now, I opened this issue as I used array_view for both mutable and immutable use there, but I suppose span could be a good replacement.

closed 2020-09-28 by bku-sue