ETLCPP/etl #265: how to include iostream correctly

mttbx, 2020-09-15:
This is not a bug report.
I like etl very much, especially its framework. But I noticed the lack of iostream.
Here's the question:
I use my own printf to fit my memory of the target board. How should I do it with cout? 
When I include iostream in my project, the compiler said the .text section is out of bound.
I noticed that it makes use of the default malloc function in the library.
So since etl don't write its own iostream, there's must some sort of  solution to this kind of problem.
Can you help me?

jwellbelove, 2020-09-15:
In one of my projects back in the early 1990s I had to write my own `printf `due to the C library using heap resources.

The best way may be to create your own `cout` implementation that utilises your custom `printf`.
You would have to overload `operator>>` for all of the C++ primitives. 
You would also have to implement the formatting options, though you could probably use the ETL's `string_stream` and `format_spec` classes to do that for you. 
Actually, if you use the ETL's formatting, you will only need to call a `puts()` implementation.

It's an interesting idea. I may look at adding an `etl::cout` that hooks into an existing `puts()` implementation.

jwellbelove, 2020-09-15:
It would have to be something other than `puts()` as it add a linefeed.

closed 2020-09-17 by mttbx