ETLCPP/etl #266: ETL migration from 15.4.2 to 18.14.1

Tarik06, 2020-09-15:
Compiler : GNU ARM GCC 9-2019-q4-major
CPP version : c++14

Hello,
My project was working fine with the ETL (15.4.2), i would like to megrate the ETL to it latest version (18.14.1).
But when i replace the include folder in my project, it doesn't compile anymore !!
A lot of errors like :
```
ell/algorithm.h:2159:3: error: ‘ETL_NODISCARD’ does not name a type
error: ‘pair’ in namespace ‘std’ does not name a template type
etl/utility.h:116:5: error: ‘ETL_CONSTEXPR’ does not name a type
  116 |     ETL_CONSTEXPR pair()
etl/utility.h:123:5: error: ‘ETL_CONSTEXPR14’ does not name a type
etl/iterator.h:70:13: error: ‘ETL_OR_STD’ does not name a type
etl/iterator.h:273:5: error: ‘ETL_CONSTEXPR17’ does not name a type
  273 |     ETL_CONSTEXPR17 reverse_iterator(const reverse_iterator<TOther>& other)
etl/integral_limits.h:136:35: error: ‘CHAR_BIT’ was not declared in this scope
etl/memory.h:418:27: error: ‘distance’ is not a member of ‘etl’; did you mean ‘std::distance’?
  418 |     count += int32_t(etl::distance(i_begin, i_end));
etl/private/pvoidvector.h:625:47: error: 'equal' is not a member of 'etl'; did you mean 'equal_to'?
  625 |     return (lhs.size() == rhs.size()) && etl::equal(lhs.begin(), lhs.end(), rhs.begin());

```

The only definition i used in the Makefile is 
CXXFLAGS += -DPROFILE_ARM_V7_GENERIC

My question:
What are the steps to migrate my project as there have been a lot of changes since 15.4.2 ?
Thank you

jwellbelove, 2020-09-15:
I didn't create the CMakeLists.txt in the etl/ directory and I don't even use it. It's a very crude way to include a profile and I don't like how it's setup, but some people use it, so it has to stay, unfortunately. 

I'm not sure that PROFILE_ARM_V7_GENERIC actually does anything at all! I think the makefile used to include an etl_profile.h in the profiles directory, but I think that disappeared a long time ago.

The profiles in the /profile directory were only ever supposed to be 'examples'; you were supposed to create your own to suit your platform.

That said, platform.h can now determine most compiler setups automatically, unless your etl_profile.h overrides them.

If you are using GCC and your compiler switches are set to C++14 then plaform.h should define the following macros...
ETL_COMPILER_GCC
ETL_CPP11_SUPPORTED 1
ETL_CPP14_SUPPORTED 1
ETL_CPP17_SUPPORTED 0
ETL_CPP20_SUPPORTED 0
ETL_COMPILER_VERSION      __GNUC__
ETL_COMPILER_FULL_VERSION ((__GNUC__ * 10000) + (__GNUC_MINOR__ * 100) + __GNUC_PATCHLEVEL__)

If you are using the STL and the compiler supports the preprocessor directive __has_include then you do not even have to define an etl_profile.h at all.

platform.h should define all of the macros that you see in your error list.

Tarik06, 2020-09-16:
after adding the definitions, it seems that it works fine with GNU GCC under linux but not with the ARM toochain (gnu_arm_gcc_9-2019-q4-major), i have the same problems !!

jwellbelove, 2020-09-16:
"after adding the definitions"
platform.h should define all of these for you.

You mean the exact same errors?
platform.h should defined ETL_NODISCARD and all the others.

Tarik06, 2020-09-16:
Yes exactly the same errors,  

```
`rror: ‘pair’ in namespace ‘std’ does not name a template type
etl/utility.h:116:5: error: ‘ETL_CONSTEXPR’ does not name a type
`etl/private/pvoidvector.h:625:47: error: 'equal' is not a member of 'etl'; did you mean 'equal_to'?
etl/utility.h:123:5: error: ‘ETL_CONSTEXPR14’ does not name a type
etl/iterator.h:70:13: error: ‘ETL_OR_STD’ does not name a type
etl/iterator.h:273:5: error: ‘ETL_CONSTEXPR17’ does not name a type
  273 |     ETL_CONSTEXPR17 reverse_iterator(const reverse_iterator<TOther>& other)

```
I will try to make a simple example with  just main.cpp that codes etl::vector<int,5> with the ARM GCC toolchain compiler (gnu_arm_gcc_9-2019-q4-major) under linux that you can see the errors by yourself, i will attach the example here

Tarik06, 2020-09-16:
Here is the example : 

[](https://polestar31-my.sharepoint.com/:u:/g/personal/tchalal_polestar31_onmicrosoft_com/EYM3Szn4LpBFvhZSlDS6I3EB7H92OhwNG985f7KoOXT-AA?e=gkl1tu)

you have just to execute "make" and then see the errors
Thanks

jwellbelove, 2020-09-16:
OK, I don't have a Linux machine but I can spin up my Linux VM and download it.

jwellbelove, 2020-09-16:
Well, I copied the example into my Linux VM and typed make and I got

john@ubuntu:/mnt/hgfs/Programming/GitHub/tchalal/etl_arm_example$ make
make: /home/tchalal/Documents/lib/etl_arm_example/gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc: Command not found
Cannot find: '/home/tchalal/Documents/lib/etl_arm_example/gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc'.
Please set values in: "/mnt/hgfs/Programming/GitHub/tchalal/etl_arm_example/Makefile.posix"
according to the actual configuration of your system.
Makefile.common:130: *** Cannot continue. Stop.
john@ubuntu:/mnt/hgfs/Programming/GitHub/tchalal/etl_arm_example$

jwellbelove, 2020-09-16:
I tried loading it into my Linux VM and typed 'make' and I got

```
john@ubuntu:/mnt/hgfs/Programming/GitHub/tchalal/etl_arm_example$ make
make: /home/tchalal/Documents/lib/etl_arm_example/gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc: Command not found
Cannot find: '/home/tchalal/Documents/lib/etl_arm_example/gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc'.
Please set values in: "/mnt/hgfs/Programming/GitHub/tchalal/etl_arm_example/Makefile.posix"
according to the actual configuration of your system.
Makefile.common:130: *** Cannot continue. Stop.
```

Tarik06, 2020-09-16:
Sorry, could you please change the "GNU_INSTALL_ROOT" in Makefile.posix to this value:
GNU_INSTALL_ROOT ?=./gnu_arm_gcc_9-2019-q4-major-linux/bin/

and then retry.
Thank you

jwellbelove, 2020-09-16:
Sorry if I'm a bit slow with Linux. I've been programming for nearly 40 years, but only two of those have been on a Linux development platform.

jwellbelove, 2020-09-16:
No joy

```
john@ubuntu:/mnt/hgfs/Programming/GitHub/tchalal/etl_arm_example$ make
make: ./gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc: Command not found
Cannot find: './gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc'.
Please set values in: "/mnt/hgfs/Programming/GitHub/tchalal/etl_arm_example/Makefile.posix"
according to the actual configuration of your system.
Makefile.common:130: *** Cannot continue. Stop.
```

jwellbelove, 2020-09-16:
```
john@ubuntu:/mnt/hgfs/Programming/GitHub/tchalal/etl_arm_example$ cat Makefile.posix
GNU_INSTALL_ROOT ?= ./gnu_arm_gcc_9-2019-q4-major-linux/bin/
GNU_VERSION ?= 9.2.1
GNU_PREFIX ?= arm-none-eabi
john@ubuntu:/mnt/hgfs/Programming/GitHub/tchalal/etl_arm_example$ 
```

jwellbelove, 2020-09-16:
I only see 
ar  as  ld  ld.bfd  nm  objcopy  objdump  ranlib  readelf  strip
in the bin directory

jwellbelove, 2020-09-16:
OK, although I downloaded the zip, I don't seem to have got all of the files. I'll try again.

jwellbelove, 2020-09-16:
john@ubuntu:/mnt/hgfs/Programming/GitHub/tchalal/etl_arm_example$ make
./gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc: 1: ./gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc: cannot create �8@@x�@8: Directory nonexistent
./gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc: 1: ./gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc: ELF: not found
./gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc: 1: ./gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc: Syntax error: Unterminated quoted string
Cannot find: './gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc'.
Please set values in: "/mnt/hgfs/Programming/GitHub/tchalal/etl_arm_example/Makefile.posix"
according to the actual configuration of your system.
Makefile.common:130: *** Cannot continue. Stop.

Tarik06, 2020-09-16:
is it Windows WSL or a VM? i just lunched my VM too and i haven't your issue !!!
[image]

Tarik06, 2020-09-16:
I just tested it in WSL and i haven't your issue too !!
Please make sure that the folder "gnu_arm_gcc_9-2019-q4-major-linux" is in the same directory as the Makefile
[image]

jwellbelove, 2020-09-16:
It may be because I'm using a shared directory between Windows and Linux. I'll copy the files over to a native Linux directory to see if that helps. I don't normally have a problem with GCC on Linux and shared directories.

jwellbelove, 2020-09-16:
Nope, same thing.

rm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc: 1: ./gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc: ELF: not found
./gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc: 1: ./gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc: Syntax error: Unterminated quoted string
Cannot find: './gnu_arm_gcc_9-2019-q4-major-linux/bin/arm-none-eabi-gcc'.
Please set values in: "/home/john/Documents/tchalal/etl_arm_example/Makefile.posix"
according to the actual configuration of your system.
Makefile.common:130: *** Cannot continue. Stop.

Tarik06, 2020-09-17:
I think that it is a privileged access issue with WSL, if you can retest it with VM, it would be great
Thank you

jwellbelove, 2020-09-17:
I have _only_ tested it with a VM so far. I plan to try WSL today.

jwellbelove, 2020-09-17:
OK, I've got it compiling with WSL.

jwellbelove, 2020-09-17:
I don't understand the error are the moment.

The first error is `./etl/utility.h:116:5: note: the macro 'ETL_CONSTEXPR' had not yet been defined`

Line 34 of `utility.h` is `#include "platform.h"`

`platform.h` will define `ETL_CONSTEXPR` as either nothing or `constexpr`

Therefore `ETL_CONSTEXPR` should _always_ be defined!

jwellbelove, 2020-09-17:
Right, I've got it to compile.
(Apart from the linking not working, I'd like to know why?)

```
ubuntu@John-PC:/mnt/d/users/John/Documents/Programming/tchalal/etl_arm_example$ make
Compiling file: main.cpp
Linking target: _build/test.out
make[1]: dump.mk: No such file or directory
make[1]: *** No rule to make target 'dump.mk'.  Stop.
make: *** [Makefile.common:294: _build/test.out] Error 2
```

I changed the Make file include directives and changed `#include "vector.h"` to `#include "etl/vector.h"`

```
ETL_LIB_DIR := ./

# Source files common to all targets
SRC_FILES += \
  ./main.cpp \

# Include folders common to all targets
INC_FOLDERS += \
  $(ETL_LIB_DIR)
```

jwellbelove, 2020-09-17:
You can also remove `CXXFLAGS += -DPROFILE_ARM_V7_GENERIC`

jwellbelove, 2020-09-17:
I suspect the issue was due to clashing header file names in the include paths.

jwellbelove, 2020-09-17:
Still doesn't work in my VM though. Is the GCC 64 bit? My VM is Ubuntu 32bit.

Tarik06, 2020-09-17:
Well now it works, thank you a lot.
Yes, it's seems that there is a collision between ARM GCC headers and ETL headers, but in the previous version (15.4), i don't have this issue !!! Anyway, the issue is resolved thanks a lot.

The toolchain ARM GCC is a 64bit for linux !

closed 2020-09-22 by jwellbelove