ETLCPP/etl #269: lto1: fatal error: multiple prevailing defs for 'invocation'

adfernandes, 2020-09-20:
Compiler: Segger Embedded Studio 5.10b, using the ARM GCC toolchain `9-2020-q2-update`.

Optimization flags: `-O1` with LTO.

`#define ETL_NO_CHECKS 1`

Result:

```
"/Applications/SEGGER Embedded Studio for ARM 5.10b/bin/mkresolution" /project/ses/ses/../_build/Sensor/Debug/Objects/Sensor.ind /project/ses/ses/../_build/Sensor/Debug/Objects/Sensor.res /project/ses/ses/../_build/Sensor/Debug/Objects/Sensor_link.ind /project/ses/ses/../_build/Sensor/Debug/Objects/Sensor_lto.ind /project/ses/ses/../_build/Sensor/Debug/Objects/Sensor_lto.txt _vectors

"/Applications/SEGGER Embedded Studio for ARM 5.10b/gcc/arm-none-eabi/bin/lto1" -quiet -mcpu=cortex-m4 -mlittle-endian -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mtp=soft -munaligned-access -O1 -fomit-frame-pointer -fno-dwarf2-cfi-asm -fno-builtin -fno-math-errno -ffunction-sections -fdata-sections -fshort-enums -fno-common -g0 -flto-partition=none -fresolution=/project/ses/ses/../_build/Sensor/Debug/Objects/Sensor.res -flinker-output=exec @/project/ses/ses/../_build/Sensor/Debug/Objects/Sensor_lto.ind -o /project/ses/ses/../_build/Sensor/Debug/Objects/Sensor_lto.s

lto1: fatal error: multiple prevailing defs for 'invocation'
```

Looking at the `res` file (after running through `c++filt`:
```
_build/Sensor/Debug/Objects/etl.o 6
249 8a76c6af17f9af59 PREVAILING_DEF_IRONLY guard variable for etl::private_error_handler::wrapper<void>::invocation
274 8a76c6af17f9af59 PREVAILING_DEF_IRONLY etl::private_error_handler::wrapper<void>::invocation

_build/Sensor/Debug/Objects/main.o 18
306 c085282439317518 PREVAILING_DEF_IRONLY guard variable for etl::private_error_handler::wrapper<void>::invocation
328 c085282439317518 PREVAILING_DEF_IRONLY etl::private_error_handler::wrapper<void>::invocation

_build/Sensor/Debug/Objects/bluetooth.o 11
350 6240df6d1cb86934 PREVAILING_DEF_IRONLY guard variable for etl::private_error_handler::wrapper<void>::invocation
383 6240df6d1cb86934 PREVAILING_DEF_IRONLY etl::private_error_handler::wrapper<void>::invocation

_build/Sensor/Debug/Objects/board.o 19
706 cb851f170b22dac3 PREVAILING_DEF_IRONLY guard variable for etl::private_error_handler::wrapper<void>::invocation
737 cb851f170b22dac3 PREVAILING_DEF_IRONLY etl::private_error_handler::wrapper<void>::invocation
```

... and so on.

**Compilation is fine if I disable LTO.**

I _think_ this is a similar bug to [this](https://sourceware.org/bugzilla/show_bug.cgi?id=23350) which references [this](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=86490), but I'm not sure.

Any simple workaround? Unless I'm mistaken, this is coming from ``etl::private_error_handler` in `error_handler.h`.

jwellbelove, 2020-09-21:
Mmm, I'm not sure what to try yet.
`etl::private_error_handler::wrapper` is workaround there to allow a static instance of `invocation_element` to exist within a header, so that a cpp file does not have to be created.

jwellbelove, 2020-09-21:
I have an idea. I'll try it out and post the code changes for you to try out.

jwellbelove, 2020-09-21:
Can you replace error.handler.h with the one in this zip file?

[error_handler.zip](https://github.com/ETLCPP/etl/files/5254336/error_handler.zip)

adfernandes, 2020-09-21:
Yes, that seems to work, both with `debug` and `release` builds, and under various `ETL_LOG_*` settings!

adfernandes, 2020-09-21:
BTW - in your opinion, who/where is the upstream bug? It's pretty clear to me that something odd is going on in the way Segger is using the GNU toolchain and LTO, but... it _should_ work, and your original code definitely was both in the spirit and the law of C++ template-de-duplication...

jwellbelove, 2020-09-21:
It definitely seems to be a bug with Segger. It seems to be seeing multiple definitions of some symbols where there are (or should be) just one.

adfernandes, 2020-09-22:
Yeah, spent some time this evening messing about with the Segger Embedded Studio compiler driver and it seems that they handle their own intermediate output for LTO, hence them using their custom `mkresolution` program.

The `mkresolution` program appears to be treating multiply-defined symbols as standard C symbols rather than (perfectly valid) multiple template definitions, and therefore (incorrectly) issues a `multiple prevailing defs` error, which would be correct for C but incorrect for C++ templates.

jwellbelove, 2020-09-22:
I actually prefer the modified code now. I think it's simpler to understand than the template wrapper method.

closed 2020-09-22 by jwellbelove

adfernandes, 2020-12-28:
Just a note if anyone finds this issue via search, Segger has apparently fixed this bug in Embedded Studio 5.32a (but I have not personally tested it).

jwellbelove, 2020-12-28:
Thanks for the update.