ETLCPP/etl #270: Question: Disabling private error handlers to conserve RAM

shreyasbharath, 2020-09-25:
Hello library maintainers,

Thanks for an awesome library for us embedded C++ folks.

I am noticing a lot of private error handlers being defined in anything that explicitly or implicitly #includes `etl/delegate.h` in the target map (ARM GCC) file. I have defined `ETL_NO_CHECKS` so I'd expect to pay no cost in terms of RAM.

```
_ZGVN3etl21private_error_handler7wrapperIvE10invocationE (4 bytes)
_ZN3etl21private_error_handler7wrapperIvE10invocationE (8 bytes)
```

How do I compile these away from my final binary?

jwellbelove, 2020-09-25:
Thanks for the heads up on that.
I probably need to add some conditional compilation macros around that.
I'll take a look this evening.

jwellbelove, 2020-09-26:
Fixed in 18.15.4

closed 2020-09-26 by jwellbelove