ETLCPP/etl #276: list::full() and list::available() assert for super empty list

bolry, 2020-09-27:
`list::full` and `list::available` should not assert for empty list with **MAX_SIZE_ == 0**. The list is logically both full and empty at the same time.
```c++
#include <cassert>
#include <etl/list.h>

int main() {
  etl::list<int, 0> x;
  etl::list<int, 0> const& cx = x;

  assert(cx.size() == 0U);
  assert(cx.max_size() == 0U);
  assert(cx.empty());
  assert(cx.full()); // ETL_ASSERT(p_node_pool != ETL_NULLPTR, ETL_ERROR(list_no_pool));
  assert(cx.available() == 0U); // ETL_ASSERT(p_node_pool != ETL_NULLPTR, ETL_ERROR(list_no_pool));

  return 0;
}
```

jwellbelove, 2020-09-27:
It is not asserting on an empty list, it is asserting on a list with an external pool, that hasn't been given a pool yet.

A size of zero indicates that the container's size is not set by the template parameter, but by the size of the supplied pool.

Your code is declaring an etl::list with no storage, so it asserts when you try to interrogate its size. If you don't supply a pool in the constructor then you must supply on later before you use it.

External pools allow storage to be defined separately from the container. It also allows pools to be shared between containers.

jwellbelove, 2020-09-27:
See [https://www.etlcpp.com/list.html]()

closed 2020-10-10 by jwellbelove