ETLCPP/etl #278: fsm framework call _sbrk indirectly

mttbx, 2020-09-29:
I'm not sure, but when I instantitaze sub class of fsm, I got an error:
```
libnosys.a(sbrk.o): in function `_sbrk':
sbrk.c:(.text._sbrk+0x18): undefined reference to `end'
```
It seems etl fsm or message framework call some stl functions and eventually call malloc.
I implement *operator new, and I can call new without any errors. And I link it with -specs=nano.specs -specs=nosys.specs.

jwellbelove, 2020-09-29:
Are you able to use the ETL_NO_STL option?
Or failing that, are you able to look at the call stack to see which STL function is calling malloc?

mttbx, 2020-09-30:
I tried to debug it, but when my program was running, no function called _sbrk. I tried to add ETL_NO_STL option in etl_profile.h, but still I got this error. Here's my content of  etl_profile.h:
```
#define ETL_TARGET_DEVICE_ARM
#define ETL_TARGET_OS_NONE
#define ETL_CPP11_SUPPORTED 1
#define ETL_CPP14_SUPPORTED 1
#define ETL_CPP17_SUPPORTED 0
#define ETL_NO_NULLPTR_SUPPORT 0
#define ETL_NO_LARGE_CHAR_SUPPORT 0
#define ETL_CPP11_TYPE_TRAITS_IS_TRIVIAL_SUPPORTED 1
#define ETL_NO_STL
```

jwellbelove, 2020-09-30:
When ETL_NO_STL is defined then the ETL should not use the STL at all.

Is it the constructor of the etl::fsm derived class that errors, or is it later on when you try to pass an event message?

mttbx, 2020-10-08:
Sorry to respond so late. It's the constructor. In fact, when I create a variable of that derived class, I get the error.

mttbx, 2020-10-09:
```
struct StateId
{
  enum
  {
    CONNECT,
    RUN
  };
};
class EventX : public etl::message<0>{};
const etl::message_router_id_t MOTOR_CONTROL = 0;

class Fsm
  : public etl::fsm
{
public:
  Fsm() : etl::fsm(MOTOR_CONTROL) {}

};

class ConnState
  : public etl::fsm_state<Fsm, ConnState, StateId::CONNECT, EventX>
{
public:
  ConnState() {}

  etl::fsm_state_id_t on_event(etl::imessage_router& sender, const EventX& event)
  {
    return STATE_ID;
  }
  etl::fsm_state_id_t on_event_unknown(etl::imessage_router& sender, const etl::imessage& event)
  {
    return STATE_ID;
  }
};

Fsm ctx;
```

```
void *operator new(size_t size)
{
    return self_malloc(size);
}

void *operator new[](size_t size)
{
    return self_malloc(size);
}

void operator delete(void *ptr)
{
    self_free(ptr);
}

void operator delete[](void *ptr)
{
    return self_free(ptr);
}
```

jwellbelove, 2020-10-09:
I tried your code under Visual Studio and GCC and it doesn't hit the `new` operator overloads for me.
The only way I van get it to hit the break points is to `new` the `Fsm` object.

What compiler are you using?

jwellbelove, 2020-10-09:
Could it be the calls to 'placement new' are being directed to 'new' on your setup?
Could you try this to see if your overloads are being called?

int i;
new (&i) int(0); // The overloaded new operator should NOT be called for this.

Also try this...
::new (&i) int(0); // The overloaded new operator should NOT be called for this.

mttbx, 2020-10-09:
I'm using arm-non-eabi-gcc toolchain.
Yes, it didn't hit my new operator. But I didn't get the warning at all this time. It seems my overloads doesn't  cover all situations.
But it's still strange that this time I didn't get the warning at all. Maybe the other new function is being called? How should I make things right?

jwellbelove, 2020-10-09:
It's difficult to know unless we can figure out where the call to `new` is coming from.
When using ETL_NO_STL then none of the STL is used in the ETL, and there are no calls to 'heap new' in the ETL.
Could your C standard library be calling `malloc`?

mttbx, 2020-10-09:
It could happen that I may misuse c function.
But if I commented out "Fsm ctx;", the warning just gone. It's hard to explain.
I just found that it called replacement new, which is defined in file new.
`inline void* operator new(std::size_t, void* __p) _GLIBCXX_USE_NOEXCEPT`
I think replacement new may not be the problem of my warning.

mttbx, 2020-10-09:
```
***/arm-none-eabi/lib/thumb/v6-m/nofp/libnosys.a(sbrk.o): in function `_sbrk':
sbrk.c:(.text._sbrk+0x18): undefined reference to `end'
```

jwellbelove, 2020-10-09:
Does this help?
https://e2e.ti.com/support/tools/ccs/f/81/t/382293?undefined-reference-to-end-in-sbrk-c-in-library-libnosys-a-

It looks like you may have to define 'end' in some way.

jwellbelove, 2020-10-09:
I also found this.

https://ww1.microchip.com/downloads/en/DeviceDoc/Frequently-Asked-Questions-4.9.3.26.txt

7. Error: undefined reference to `end'

   If you used nosys.specs or rdimon.specs, it may expect symbol 'end' to be
   defined to denote the HEAP start. Define symbol 'end'.
   Refer "12. How to update linker scripts for syscalls?" for examples.

mttbx, 2020-10-09:
Normally I don't need _sbrk at all, It won't be linked. So the real problem is why this function is being called. I am looking at map file. If I find something, I'll let you know. Thank you.

jwellbelove, 2020-10-09:
Maybe it's not being called, but it is being linked by default.

closed 2020-10-17 by jwellbelove