ETLCPP/etl #282: cppcheck 2.2 warnings

labodj, 2020-10-03:
Hi,
I've just updated [cppcheck](https://github.com/danmar/cppcheck) to 2.2 and these are the errors and warnings related to this library, for some of them I can make a pull request, but for others I can't since my c++ skills are not so good and I can't avaluate if the warning is a false positive.

I'm using it with Arduino AVR, [THIS](https://github.com/labodj/avr_stl/tree/merge-upstream) STL support library, and AVR-GCC 10.2, compiled with c++20 standard.

with this config:

```c++
#ifndef __ETL_PROFILE_H__
#define __ETL_PROFILE_H__

#define ETL_VERBOSE_ERRORS
#define ETL_CHECK_PUSH_POP
#define ETL_NO_STL
#define ETL_TARGET_DEVICE_GENERIC
#define ETL_TARGET_OS_NONE
#define ETL_COMPILER_GCC
#include "etl/profiles/auto.h"
#endif // __ETL_PROFILE_H__
```

Have a look here:

```
Embedded Template Library/include/etl/private/pvoidvector.h:227: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/pvoidvector.h:237: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/pvoidvector.h:249: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/pvoidvector.h:261: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/pvoidvector.h:270: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/pvoidvector.h:279: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/pvoidvector.h:288: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/pvoidvector.h:297: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:212: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:222: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:233: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:244: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:253: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:262: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:271: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:280: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:616: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:626: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:637: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:648: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:657: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:666: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:675: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/private/ivectorpointer.h:684: [high:error] Reference to temporary returned. [returnTempReference]
Embedded Template Library/include/etl/memory.h:1341: [medium:warning] Member variable 'unique_ptr::p' is not assigned a value in 'unique_ptr::operator='. [operatorEqVarError]
Embedded Template Library/include/etl/memory.h:1471: [medium:warning] Member variable 'unique_ptr::p' is not assigned a value in 'unique_ptr::operator='. [operatorEqVarError]
Embedded Template Library/include/etl/function.h:176: [low:style] Class 'function' has a constructor with 1 argument that is not explicit. [noExplicitConstructor]
Embedded Template Library/include/etl/function.h:209: [low:style] Class 'function' has a constructor with 1 argument that is not explicit. [noExplicitConstructor]
Embedded Template Library/include/etl/function.h:246: [low:style] Class 'function_mp' has a constructor with 1 argument that is not explicit. [noExplicitConstructor]
Embedded Template Library/include/etl/function.h:284: [low:style] Class 'function_mv' has a constructor with 1 argument that is not explicit. [noExplicitConstructor]
Embedded Template Library/include/etl/private/vector_base.h:153: [low:style] Class 'vector_base' has a constructor with 1 argument that is not explicit. [noExplicitConstructor]
Embedded Template Library/include/etl/function.h:151: [low:style] The function 'operator()' overrides a function in a base class but is not marked with a 'override' specifier. [missingOverride]
Embedded Template Library/include/etl/function.h:217: [low:style] The function 'operator()' overrides a function in a base class but is not marked with a 'override' specifier. [missingOverride]
Embedded Template Library/include/etl/function.h:293: [low:style] The function 'operator()' overrides a function in a base class but is not marked with a 'override' specifier. [missingOverride]
Embedded Template Library/include/etl/function.h:347: [low:style] The function 'operator()' overrides a function in a base class but is not marked with a 'override' specifier. [missingOverride]
Embedded Template Library/include/etl/function.h:411: [low:style] The function 'operator()' overrides a function in a base class but is not marked with a 'override' specifier. [missingOverride]
Embedded Template Library/include/etl/algorithm.h:136: [low:style] Local variable 'count' shadows outer function [shadowFunction]
Embedded Template Library/include/etl/algorithm.h:303: [low:style] Local variable 'count' shadows outer function [shadowFunction]
Embedded Template Library/include/etl/algorithm.h:357: [low:style] Local variable 'count' shadows outer function [shadowFunction]
Embedded Template Library/include/etl/algorithm.h:1843: [low:style] Local variable 'next' shadows outer function [shadowFunction]
Embedded Template Library/include/etl/algorithm.h:2559: [low:style] Local variable 'size' shadows outer function [shadowFunction]
Embedded Template Library/include/etl/algorithm.h:2713: [low:style] Local variable 'size' shadows outer function [shadowFunction]
Embedded Template Library/include/etl/map.h:205: [medium:warning] Member variable 'Node::children' is not initialized in the constructor. [uninitMemberVar]
Embedded Template Library/include/etl/pool.h:626: [medium:warning] The class 'pool < etl :: imap < const char * , uint8_t , charCmp > :: Data_Node , MAX_SIZE >' defines member variable with name 'SIZE' also defined in its parent class 'generic_pool < sizeof ( etl :: imap < const char * , uint8_t , charCmp > :: Data_Node ) , etl :: alignment_of < etl :: imap < const char * , uint8_t , charCmp > :: Data_Node > :: value , MAX_SIZE >'. [duplInheritedMember]
Embedded Template Library/include/etl/pool.h:627: [medium:warning] The class 'pool < etl :: imap < const char * , uint8_t , charCmp > :: Data_Node , MAX_SIZE >' defines member variable with name 'ALIGNMENT' also defined in its parent class 'generic_pool < sizeof ( etl :: imap < const char * , uint8_t , charCmp > :: Data_Node ) , etl :: alignment_of < etl :: imap < const char * , uint8_t , charCmp > :: Data_Node > :: value , MAX_SIZE >'. [duplInheritedMember]
Embedded Template Library/include/etl/pool.h:628: [medium:warning] The class 'pool < etl :: imap < const char * , uint8_t , charCmp > :: Data_Node , MAX_SIZE >' defines member variable with name 'TYPE_SIZE' also defined in its parent class 'generic_pool < sizeof ( etl :: imap < const char * , uint8_t , charCmp > :: Data_Node ) , etl :: alignment_of < etl :: imap < const char * , uint8_t , charCmp > :: Data_Node > :: value , MAX_SIZE >'. [duplInheritedMembe [... cut ...]
Embedded Template Library/include/etl/pool.h:628: [medium:warning] The class 'pool < etl :: imap < uint8_t , uint32_t , etl :: less < uint8_t > > :: Data_Node , MAX_SIZE >' defines member variable with name 'TYPE_SIZE' also defined in its parent class 'generic_pool < sizeof ( etl :: imap < uint8_t , uint32_t , etl :: less < uint8_t > > :: Data_Node ) , etl :: alignment_of < etl :: imap < uint8_t , uint32_t , etl :: less < uint8_t > > :: Data_Node > :: value , MAX_SIZE >'. [duplInheritedMember]
```

jwellbelove, 2020-10-03:
Thanks, that looks like some genuine errors in there.
I'll take a look.

jwellbelove, 2020-10-03:
Actually, the " reference to temporary" is not a real error, it's just getting confused over an unnecessary cast to a reference. I'll get rid of the casts anyway.

jwellbelove, 2020-10-03:
Actually, the "reference to temporary" is not correct. I think the cppcheck is getting confused with the reference cast. The cast is actually redundant as `p_buffer[I]` is a `void*` and `reference` is a reference to a `void*`, so it should be fine without the cast. Hopefully cppcheck will be fine with that too.

jwellbelove, 2020-10-03:
The other warnings should be easily fixed. The etl:: function warnings require the addition of ETL_OVERRIDE.

jwellbelove, 2020-10-04:
I've downloaded cppcheck and run it on the ETL code.
I'm getting a lot of false positives from it; for example...

```
    template <typename THash>
    size_t get_hash(parameter_t key) const
    {
      size_t hash = THash()(key);

      // Fold the hash down to fit the width.
      return fold_bits<size_t, etl::log2<WIDTH>::value>(hash); <<<< USED HERE
    }
```
This gives the warning that "Variable 'hash' is assigned a value that is never used" when it quite plainly is!

Also it does not seem to expand macros correctly, as I get a 'not marked as override' when I use `ETL_OVERRIDE` (defined as `override`) but is fine with `override`.

jwellbelove, 2020-10-04:
And "Local variable 'count' shadows outer function" appears to be triggered by this...
```
    size_t count() const
    {
      size_t n = 0;

      for (size_t i = 0; i < SIZE; ++i)
      {
        n += etl::count_bits(pdata[i]);
      }

      return n;
    }
```
I assume it is automatically expanding the inline function and producing the warning, which I believe to be incorrect.

labodj, 2020-10-04:
Hi @jwellbelove I believe infact that cppcheck results contain some false positive.
Anyway I find it useful because some of them are real, and some of them results in false positive because of the way I wrote the code that is not so "brilliant", I'm sure this is not the case, your library is very well written, but I think it's always nice to double check the code with tools like this, I'm currently using cppcheck and clangtidy to check my sources.
If you are sure you find some false positive, and looking at them it seems that you did, it could be useful to submit the bug in their chat, forum or bug tracker, [here](https://trac.cppcheck.net/)

closed 2020-10-24 by jwellbelove