ETLCPP/etl #283: etl::circular_buffer

shreyasbharath, 2020-10-08:
Hello again,

I was looking around in the documentation and I couldn't find a circular buffer implementation in ETL.

You could wrap `etl::queue` to create one, but I am not sure that'd be the most efficient implementation.

Thoughts?

jwellbelove, 2020-10-08:
Funny enough, I was looking at adding one just yesterday.

shreyasbharath, 2020-10-08:
Awesome!

shreyasbharath, 2020-10-12:
@jwellbelove On top of creating a fixed size circular buffer, we should also look at adding a circular buffer `span` or `view` like so:
https://github.com/martinmoene/ring-span-lite

The proposal to add it to the C++ standard
http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2017/p0059r3.pdf

jwellbelove, 2020-10-12:
That is also an interesting feature to add.

closed 2020-10-24 by jwellbelove

shreyasbharath, 2020-10-24:
Thanks for adding this @jwellbelove