ETLCPP/etl #285: [suggesion] Add protothreads-like spport

puzrin, 2020-10-11:
Current scheduler/task concept is too primitive for many use cases. To express long business logic, it would be useful to have `YIELD(result)`, `WAIT_UNTIL(expression)` and so on.

This approach is implemented in protothreads (and has no OS dependencies). See https://www.google.com/search?q=github+protothreads.

Can this become a part of ETL or this is out of project scope? Full-featured RTOS is overkill in many cases. ETL already has everyhing for messaging. With proothreads it will be nice & cheap alernative to RTOS-es, out of box.

jwellbelove, 2020-10-11:
Thanks for the suggestion, I'll take a look.

puzrin, 2020-10-17:
Can i help anyhow?

Here is my proposal for read:

- https://github.com/Akagi201/protothreads - classic, with modular context switcher
  - sugar for subcalls (PT_SPAWN).
  - need manually write top level loop (see samples)
- https://github.com/benhoyt/protothreads-cpp - minimalistic, for cpp
- https://github.com/LarryRuane/protothread
  - tries to reinvent scheduler an loop only active "threads"
  - but uses malloc/free.

There are much more projects, but those will not add new programming patterns.

List of "context switch" methods:

- "switch/while" macros, emulation for "goto variable" - 100% compatible, but real black magick. Will look horrible in debugger.
- "goto variable" - ideal, if compiler supports it.
- setjmp - better compatibility than "goto variable", but slow.

jwellbelove, 2020-10-17:
Thanks, I read up on those.
I'll have to try out some examples to fully grasp the issues that they are proposing to solve.

puzrin, 2020-10-17:
>  I'll have to try out some examples to fully grasp the issues that they are proposing to solve.

Those examples are good for coding, but not good to learn use cases.

Top use cases are:

- Non-blocking delay
- Non-bloking IO (read/write i2c/spi/uart)
- Wait for event, with timeout

https://github.com/speedcontrols/ac_sc_grinder/blob/master/src/calibrator/calibrator_pid.h#L64-L434 - that's example of PID calibrator in my project. The real working code, doing something more useful than led blinking :).

jwellbelove, 2022-01-23:
Moved to 'For future consideration'

closed 2022-01-23 by jwellbelove