ETLCPP/etl #287: ETL on 8-Bit Microcontrollers?

HpLightcorner, 2020-10-17:
Hey,

**Test-Environment**
Arduino 1.8.13

**Issue**
I am currently evaluating ETL as a major component for various libraries going to be used to educate students in control theory and hence shall be available across different ecosystems (platform.io, Arduino...) and for various boards (Adafruit Feather-Like, 8-Bit Arduinos, ARM-Arduinos etc.). I am planning to directly embed a version of ETL into my library package as the ETL-Installation vor Arduino is not compatible with the Arduino library folder structure. 

I was able to build my first tests successfully on ARM-Based devices, however, the 8-Bit builds are causing some questionmarks and I am quite unsure how to proceed. For the 8-Bit controllers, there are some defines missing needed in `<limits.h>`:

- HUGE_VAL
- HUGE_VALL
- HUGE_VALF

I simply created the defines in my `etl_profile.h` to mock support:
```
#pragma once

// ARM specific
#define ETL_VERBOSE_ERRORS
#define ETL_CHECK_PUSH_POP
#define ETL_TARGET_DEVICE_GENERIC
#define ETL_TARGET_OS_NONE
#define ETL_COMPILER_GCC
#define ETL_CPP11_SUPPORTED 1
#define ETL_NO_LARGE_CHAR_SUPPORT 1
#define ETL_CPP11_TYPE_TRAITS_IS_TRIVIAL_SUPPORTED 0
#define ETL_NO_STL

// 8-Bit specficic 
#define ETL_NO_CPP_NAN_SUPPORT
#ifndef HUGE_VAL
#define HUGE_VAL
#endif

#ifndef HUGE_VALF
#define HUGE_VALF
#endif

#ifndef HUGE_VALL
#define HUGE_VALL
#endif
```

Actually I am not quite sure why they are missing, it seems they are not defined in avr-g++ `math.h`. I am nut sure why they are not defined, maybe someone has an answer for me? 

**Proposed Solution**
I am quite unsure how to proceed. It seams that avr-g++ does support float for those small devices, but why are those defines Missing? I think it would be an option to make these defines based on a define, like `ETL_NO_HUGE_SUPPORT`?

HpLightcorner, 2020-10-17:
Some additional links:
[AVR-Math](https://www.nongnu.org/avr-libc/user-manual/group__avr__math.html)
[C-Reference to HUGE_VAL](https://en.cppreference.com/w/c/numeric/math/HUGE_VAL)

Seems like AVR-Math is not Standard-Conform when it comes to those defines?

HpLightcorner, 2020-10-17:
Okay, just to come up with a solution I derived from the ARM-Toolchain which actually defines the constants in `math.h`, I think this should also work as expected for the 8-Bit AVRs?

In limits.h add:
```
#if defined(ETL_NO_HUGE_SUPPORT)
 #ifndef HUGE_VAL
  #define HUGE_VAL (1.0e999999999)
 #endif

 #ifndef HUGE_VALF
  #define HUGE_VALF (1.0e999999999F)
 #endif

 #ifndef HUGE_VALL
  #define HUGE_VALL (1.0e999999999L)
 #endif
#endif
```

jwellbelove, 2020-10-17:
Thanks, I'll look to add those, unless you'd like to do a pull request?

Maybe name it as ETL_NO_HUGE_VAL_SUPPORT?

HpLightcorner, 2020-10-17:
I will do a pull request with your naming-suggestion :)

closed 2020-10-24 by jwellbelove

positron96, 2021-06-22:
Is this resolved? 

When enabling ETL_NO_HUGE_VAL_SUPPORT I get this compile error (in avrgcc for atmega328):

```
.pio\libdeps\pro16MHzatmega328\Embedded Template Library\include/etl/limits.h:548:5: error: floating constant exceeds range of 'float' [-Werror=overflow]
     static float infinity() { return HUGE_VALF; }
```

positron96, 2021-06-22:
Ok, I've fixed it with `-Wno-overflow`

jwellbelove, 2021-06-22:
Which version are you using?
ETL_NO_HUGE_VAL_SUPPORT has not existed since v20.9.0

jwellbelove, 2021-06-22:
I've just noticed that the definitions for HUGE_VAL and HUGE_VALF are swapped.
`limits.h`, lines 76 to 78 should read...

```
  #define HUGE_VALF FLT_MAX
  #define HUGE_VAL  DBL_MAX
  #define HUGE_VALL LDBL_MAX
```

positron96, 2021-06-23:
> Which version are you using?
> ETL_NO_HUGE_VAL_SUPPORT has not existed since v20.9.0

My version is 19.3.5 from Platformio, it mentions this define in `limits.h` (line 57).