ETLCPP/etl #289: Suggestions for improving overhead of etl::string?

dnmiller, 2020-10-20:
Hi, I was playing around with etl::string and was rather surprised that there seem to be around 8 to 12 words of overhead added to the class size, depending on compiler options (GCC 5.4). This seemed true regardless of the max size of the string. In fact, it was larger in memory than std::string in most cases.

Any idea what could be causing this? I noticed a lot macros in string.h and basic_string.h, but I was unable to find a combination that gave me a smaller footprint string.

jwellbelove, 2020-10-20:
`etl::basic_string`, apart from the storage for the actual characters, has only three member variables.

`size_t current_size`
`char* p_buffer`
`bool is_truncated`

I'll take a look to see if the ordering of the declarations can improve the packing.

jwellbelove, 2020-10-24:
Actually there is a fourth member `bool  clear_afer_use`

I will look at adding ETL compilation macros that allow _truncation_ and _clear_ functionality to be disabled.

jwellbelove, 2020-10-24:
It may also be worth adding an optional size type or memory model option for optimising strings when you know that they will never be larger than say 255 characters.

jwellbelove, 2020-10-24:
After a bit of experimentation, I found I can get the overhead as low as 8 bytes if _truncation_ and _clear_ functionality are disabled and the size type is set to `uint16_t`  or smaller, so probably worth doing.

dnmiller, 2020-10-25:
Wow, thanks for looking into this. After some more thought, I decided a plain std::array of chars was probably better suited for my application, since I know the range of everything stored at compile time.

In some sense, if you're controlling the storage up front, then isn't the char pointer and the size variable redundant?

jwellbelove, 2020-10-25:
The size variable is required because it counts the number of characters in the text, not the size of the storage. To alleviate code bloat, the strings derive from a base class that does not have the storage size as a non-type template parameter. This allows most of the code to be shared between all strings, regardless of storage size. It also allows one parameter type to be used instead of a different type for each sized string.

```
etl::string<10> text10;
etl::string<20> text20;
etl::string<30> text30;
```

The above strings all share the same implementation from `etl::istring`, but `etl::istring` needs to be told the address of the buffer and its capacity at construction.

`etl::istring` also allows a common parameter type to be used instead of multiple types.

```
void Print(const etl::istring& text);
```

Instead of...

```
void Print(const etl::string<10>& text);
void Print(const etl::string<20>& text);
void Print(const etl::string<30>& text);
```

rolan-reznik, 2020-10-26:
Hello, 
Sorry for getting into the conversation.
I have encountered some problems with etl::string memory usage also and also ended up by using array in some cases. Maybe it is possible to use bit flags for `is_truncated` and `clear_after_use`, then those variables can be squashed into on byte?

jwellbelove, 2020-10-26:
Bit flags are what I working on at the moment.

rolan-reznik, 2020-10-26:
Great. Thanks for the reply.

jwellbelove, 2020-10-26:
Adding a user defined size type is a bit more involved as a decision would have to be made as to whether the user size type is just used internally (keeping the size_t in the public API), or using the user size type in the interface too (this creates a ton of implicit cast warnings in the unit tests, and would similar issues in user code. Either way, a lot of static cast will need to be added keep the compiler happy.

rolan-reznik, 2020-10-26:
Also there could be problems with integer promotions if size type is less than word size on target platform, and it will also generate alot of warnings.

jwellbelove, 2020-10-26:
The the downside of using an etl::array as a string type is that you have no choice but to pass it around as 'pointer + size', negating the usefulness of a container class, unless all of your strings have identical max size.

jwellbelove, 2020-10-30:
The only way to make `etl::string` _really_ small would be to have an option that removed the _count_ and max _size_ members and made `size()` an O(n) operation and have `max_size()` disabled or return a maximum value of size_t.

jwellbelove, 2020-11-05:
The overhead for strings have been reduced in the latest release.
There are macro options for removing the truncation and clear-on-use flags from the class.

closed 2020-11-05 by jwellbelove